espressif/esp-csi #100: Got unexpected CSI data with CSI_RECV_ROUTER demo

Sharemon, 2023-04-04:
Hello，

    When I used the **csi_recv_router** demo to get csi data (the router is a commercial wifi router), I got some poblems about the csi data. I'd appreciate it if you could offer some help on it.

    1. when I use the demo, I got some outliers in csi data. The the csi amplitude waveform of some subcarrier is shown below. And I found the outliers is much more when I lower the ping frequency in demo, so I wonder if the outlier is normal or not. 
[image]

    2. As in my application, I need get csi data in a fixed frequency. But there was always some extra csi data even thought I set the ping frequency fixed (with the macro CONFIG_SEND_FREQUENCY). Also I tried to comment wifi_ping_router_start() in app_main() as to set the demo not to ping router, but I can still receive some csi data from uart. I wonder where the unexpected csi data comes from.  And How could I shield such csi data?

Thanks so much,
Sharemon

kszlgithub, 2023-04-04:
For the first question, we didn't see where the outliers are in the waveform. Could you please draw more cycles (and provide the frequency of pinging and the device you used) ?

For the second question, what was the rate at which you were sending pings? The ESP32 series can send pings at quite high frequency, but typical routers respond to ping commands at a rate of about 100-300Hz, so this could be a router issue.

> Hello，
>
> When I used the **csi_recv_router** demo to get csi data (the router is a commercial wifi router), I got some poblems about the csi data. I'd appreciate it if you could offer some help on [...]

Sharemon, 2023-04-06:
[image]
The outliers are hilighted as above. And the frequency of ping I used is 15Hz. Also I test 3 other routers, the outliers and unexcept csi data are still there.

I have done more invetigation into the outliers and find the outlier in question 1 and unexcept csi data in question 2 is due to  the same reason.

When I look at the rx ctrl info captured in wifi_csi_rx_cb(), I found the sig_mode of outlier frame is 0, while the sig_mode of normal csi data is 1.  Although I don't know what caused the difference, I could get perfectly normal csi data when I remove the csi data with sig_mode = 0.

zhanzhaocheng, 2023-05-31:
@Sharemon Sorry no one got back to you earlier.

Your problem is caused by the difference in transmit power between 11g and 11n
- **sig_mode = 0:** non HT(11bg) packet;
- **sig_mode = 1:**  HT(11n) packet;

Sharemon, 2023-05-31:
Thanks for your reply. I think this issue could be closed with your explaination. Thanks again.

closed 2023-05-31 by Sharemon