espressif/esp-csi #101: Questions about console_test routines collecting data

catcatBlue, 2023-04-04:
## Environment

**IDF version**: ESP-IDF v4.4.3-dirty
**esp-csi version**: [master](https://github.com/espressif/esp-csi/tree/854fbe1517857a241533d7e989b7f56a8810d9b2), commit: 854fbe1517857a241533d7e989b7f56a8810d9b2
**Operating System**: VMware, Ubuntu20.04 LTS
**Development Kit**: ESP32-S3 \* 2(External antenna)
**Power Supply**: USB HUB with external power

## Background

I want to collect both lltf and htltf data, and I want to understand the sequence of data.
> 我想同时采集 lltf 和 htltf 数据，并且想要了解数据的顺序。

I'm trying to use two esp32-s3 and esp-csi to collect data csi data, using csi_send default configuration on the sender side and console_test on the receiver side or modified to use [esp_csi_tool.py](https://github.com/espressif/esp-csi/blob/master/examples/console_test/tools/esp_csi_tool.py) for csi_recv.
> 我正在尝试使用两个 esp32-s3 以及 esp-csi 来采集数据 csi 数据，发送端使用 csi_send 默认配置，接收端使用 console_test 或者经过修改可以使用[esp_csi_tool.py](https://github.com/espressif/esp-csi/blob/master/examples/console_test/tools/esp_csi_tool.py)的 csi_recv。

I didn't find the `esp_wifi_set_csi_config(&csi_config);` statement in the console_test routine, and in the `static void csi_data_print_task(void *arg)` it only matches "LLFT", "HT-LFT", " STBC-HT-LTF". Since "radar" is not open source, I think it is not safe to add `esp_wifi_set_csi_config` directly to capture both lltf and htltf in order to avoid unpredictable situations.
> console_test 例程中我没有找到`esp_wifi_set_csi_config(&csi_config);`语句，而且在`static void csi_data_print_task(void *arg)`中也只匹配了"LLFT"、"HT-LFT"、"STBC-HT-LTF"中的一个，由于 radar 并没有开源，为了避免无法预料的情况，我认为直接增加`esp_wifi_set_csi_config`以同时采集采集 lltf 和 htltf 的做法并不保险。

So I modified csi_recv to be able to capture lltf and htltf data and to use [esp_csi_tool.py](https://github.com/espressif/esp-csi/blob/master/examples/console_test/tools/esp_csi_tool.py) for the purpose of capturing lltf and htltf data and using [esp_csi_tool.py](https://github.com/espressif/esp-csi/blob/master/examples/console_test/tools/esp_csi_tool.py). And it also achieves the purpose that the results corresponding to the total number of bytes in the [table](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-guides/wifi.html#id56) can be correctly obtained according to the configuration.
> 因此我修改了 csi_recv 以达到能采集 lltf 和 htltf 数据以及使用[esp_csi_tool.py](https://github.com/espressif/esp-csi/blob/master/examples/console_test/tools/esp_csi_tool.py)的目的。并且也达成了目的，可以根据配置正确地获取到与[表格](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-guides/wifi.html#id56)中的总字节数对应的结果。

In order to understand the structure of `buf` in `wifi_csi_info_t`, I made the following table, the table on the left is drawn based on the data collected when `lltf_en: 1, htltf_en:0, stbc_htltf2_en: 0`, the table in the middle corresponds to the configuration of `lltf_en: 0. htltf_en:1, stbc_htltf2_en: 0`, and the table on the right shows the channel and packet information set up when I collected.
> 为了方便理解`wifi_csi_info_t`中的`buf`的结构，我制作了下面这张表格，左侧的表格是根据`lltf_en: 1, htltf_en:0, stbc_htltf2_en: 0`时采集的数据所绘制的，中间的表格的对应的配置是`lltf_en: 0, htltf_en:1, stbc_htltf2_en: 0`，右侧的表格是我采集时的设置信道和数据包信息的情况。

[image]

where `Buffer index` corresponds to the index of `buf` in `wifi_csi_info_t` and `Sub-carrier index` is from [table](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-guides/wifi.html#id56).
> 其中，`Buffer index`对应的是`wifi_csi_info_t`中的`buf`的索引，`Sub-carrier index`来自[表格](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-guides/wifi.html#id56)。

## Question

1. Is the table I drew correct? Please point out if there are errors.
   > 我绘制的表格是否正确？如果有错误请指出。

1. What data is captured by console_test?
   > console_test 采集了哪些数据？

   The console_test routine has this sentence `info->valid_len = info->valid_llft_len + info->valid_ht_lft_len;`, I didn't find the source of `valid_llft_len` and `valid_ht_lft_len`. Then I set `LLFT` and `HT-LFT` by `radar --csi_output_type` command, and I found that the length of the data collected after setting `LLFT` is 104 which corresponds to 52 in the left side of the table I made, but when setting `HT-LFT` the length of the data collected is 160 which does not correspond to the table I made. The values of `valid_llft_len` and `valid_ht_lft_len` are as follows (I found the same result for len after setting `LLFT` or `HT-LFT`), how are these determined?
   > console_test 例程中有这么一句`info->valid_len = info->valid_llft_len + info->valid_ht_lft_len;`，我没有找到`valid_llft_len`和`valid_ht_lft_len`的来源。随后通过`radar --csi_output_type`命令分别设置了`LLFT`以及`HT-LFT`，我发现设置`LLFT`后采集的数据的长度为 104 与我制作的表格左侧中的 52 是对应的，但是当设置`HT-LFT`后采集的数据的长度为 160，与我制作的表格不能对应。`valid_llft_len`和`valid_ht_lft_len`的数值如下（我发现设置`LLFT`或`HT-LFT`后 len 的数值结果一样），请问这些是怎么确定的？

   ```
   I (24797) app_main: valid_llft_len = 104
   I (24798) app_main: valid_ht_lft_len = 56
   I (24798) app_main: valid_stbc_ht_lft_len = 0
   ```

   In particular, ht_lft, if the table I plotted is correct, what data is `HT-LFT` discarding? I need to deal with phase information, so the subcarrier index order is important to me (if it's not convenient to answer you can ignore it, I can use the order of the table I made).
   > 特别是 ht_lft，如果我绘制的表格是正确的，那么`HT-LFT`舍弃了哪些数据？我需要处理相位信息，因此子载波索引顺序对我很重要（如果不方便回答可以忽略，我可以使用我制作的表格里的顺序）。

Finally, thank you to all the contributors. Love from China.
> 感谢各位贡献者。爱来自瓷器。

xiangshang-kok, 2023-04-05:
你好，我想问一下，运行esp_csi_tool.py是在哪个目录下呢？
[image]
我在这个esp-csi/examples/console_test/tools目录下运行，能显示出可视化界面，但是无法显示出来波形，
我的设备是两个ESP32

kszlgithub, 2023-04-05:
@xiangshang-kok Your esp_csi_tool.py is running in the correct path. May I ask what program is burned for your CSI receiver? The CSI receiver should burn /examples/console_test instead of /examples/get_started/csi_recv.

xiangshang-kok, 2023-04-05:
@kszlgithub Yes，sir,I burned /examples/console_test for my CSI receiver,but I still cannot get info on my UI.
In addition, I would like to ask, is the CSI_send only power on the line, do not need to start through the command line

kszlgithub, 2023-04-05:
@xiangshang-kok Is the serial port set correctly when you run esp_csi_tool.py then? And yes, after buring the csi_send device, what you need to do is to power it on.

catcatBlue, 2023-04-05:
> @kszlgithub Yes，sir,I burned /examples/console_test for my CSI receiver,but I still cannot get info on my UI. In addition, I would like to ask, is the CSI_send only power on the line, do not need to start through the command line

## 1
> 运行 esp_csi_tool.py 是在哪个目录下呢？

I also run it in the directory `esp-csi/examples/console_test/tools`.
*我也是在`esp-csi/examples/console_test/tools`这个目录下运行。*

## 2
> In addition, I would like to ask, is the CSI_send only power on the line, do not need to start through the command line

Yes, console_test turns on csi by default and will send data to the uplink as soon as the receiver receives the signal. As kszlgithub said, just power on the esp32 running csi_send, then connect the esp32 running console_test to the pc, and then run `python esp_csi_tool.py -p /dev/ttyXXX` according to the corresponding serial port number.
*是的，console_test 默认开启 csi，只要接收端收到了信号就会向上位机发送数据。如 kszlgithub 所说，只需要打开运行 csi_send 的 esp32 的电源，然后连接运行 console_test 的 esp32 到 pc 上，再根据对应的串口号运行`python esp_csi_tool.py -p /dev/ttyXXX`即可。*

## Also add some

After opening the UI, please check the "log" in the bottom left corner to see if there is any information output, if not, it may be the serial port number or baud rate is not correct, you need to check it by yourself.
*打开 UI 后，请检查左下角的"log"有没有信息输出，如果没有可能是串口号或者波特率不对，你需要自行检查。*

If there is output, for example:
*如果有输出，例如：*

```
I (704) wifi_init: WiFi RX IRAM OP enabled
I (705) phy_init: phy_version 503,13653eb,Jun 1 2022,17:47:08
I (737) wifi:mode : null
I (738) wifi:Set ps type: 0
I (738) wifi:ic_enable_sniffe
I (739) esp_radar: CSI_RADAR Version: 0.0.7, Compile time: Dec 9 2022 - 17:22:21
I (740) esp_radar: csi_recv_interval: 10, csi_handle_time: 250, csi_handle_window: 50, csi_handle_buffer: 60
```

And there is no output after that, probably because the request from csi_send was not received, one possibility is that you commented out `#define RECV_ESPNOW_CSI` in console_test, resulting in `esp_wifi_set_promiscuous(true);` not being executed and not receiving the esp-now information.
*并且之后就没有输出了，可能是因为没接收到来自 csi_send 的请求，一种可能是你注释掉了 console_test 中的`#define RECV_ESPNOW_CSI`，导致`esp_wifi_set_promiscuous(true);`没有执行，无法接收到 esp-now 信息。*

Also, if your esp32 is an ESP32-S3 and you're using a usb analog serial port (such as esp-s3-box using a usb serial port), You need to select `USB Serial/JTAG` from `(Top) → Component config → ESP System Settings → Channel for console output` in `menuconfig`, and then the following code:
*此外，如果你的 esp32 是 esp32-s3，并且使用的是 usb 模拟的串口（比如 esp-s3-box 使用 usb 串口），你需要在`menuconfig` 中的`(Top) → Component config → ESP System Settings → Channel for console output`中选择`USB Serial/JTAG Controller`，然后将以下代码：*

https://github.com/espressif/esp-csi/blob/854fbe1517857a241533d7e989b7f56a8810d9b2/examples/console_test/main/app_main.c#L537-L541

Modify to:

```c
    esp_console_repl_t *repl = NULL;
    esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
    repl_config.prompt = "csi>";
#if defined(CONFIG_ESP_CONSOLE_UART_DEFAULT) || defined(CONFIG_ESP_CONSOLE_UART_CUSTOM)
    esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
#elif defined(CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG)
    esp_console_dev_usb_serial_jtag_config_t hw_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&hw_config, &repl_config, &repl));
#else
#error Unsupported console type
#endif
```

xiangshang-kok, 2023-04-05:
Thanks ALL.The problem was that my Anconda environment configuration was affecting the environment variables, causing the UI interface to come out but not receive data. After uninstalling Anconda, I reinstalled idf and recompiled the build project and it was fine! THANKS!

kszlgithub, 2023-04-12:
Hi,@catcatBlue. 
For your first question, the table you drew is correct. 
For your second question, with your help, we have also find the same issue. We are working on it and trying to find out where the problem at. Thanks for your contributions.

catcatBlue, 2023-04-12:
> Hi,@catcatBlue. For your first question, the table you drew is correct. For your second question, with your help, we have also find the same issue. We are working on it and trying to find out where the problem at. Thanks for your contributions.

Thanks, I'm already working on the acquired csi data now. Figure 1 and Figure 2 are the images of csi amplitude and phase plotted with the esp32s3 csi data, Figure 1 is for lltf and Figure 2 is for htltf, I find that the htltf data is usually more sensitive. Figure 3 shows the 3d image of lltf.

*谢谢，我现在已经在处理采集的csi数据了。图1和图2是用esp32s3采集csi数据绘制的csi幅度和相位的图像，图1是"lltf"的，图2是"htltf"的，我发现htltf的数据通常更加灵敏。图3是"lltf"的三维图像。*

[image]

<p align="center"> Figure 1</p>

[image]

<p align="center"> Figure 2</p>

[image]

<p align="center"> Figure 3</p>

But I was having a headache how to handle the outliers. You can clearly see a spike on the left side of Figure 1. I find that filtering with the hampel filter does not work well, either because the values are not right or because I don't know how to use it. Figure 4 shows the effect of filtering out the spike on the left side of Figure 1 using the hampel filter with window radius=10 and nsigma=2. However, more noise appears, and if I set nsigma=3, the noise is reduced, but the spike cannot be removed.

*但是我正在头疼异常值怎么处理。可以很明显地看到图1的左侧有一个尖峰。我发现使用hampel滤波器过滤的效果并不好，也可能是因为取值不对，或者我不会使用。图4是使用hampel滤波器过滤掉图1左侧有的尖峰的效果，窗口半径=10，nsigma=2。但是出现了更多的噪声，如果设置nsigma=3，噪声减小了，但是尖峰无法去除。*

[image]

<p align="center"> Figure 4</p>

I haven't found information about hampel filter processing complex numbers. For csi data, I use hampel filter to process the real part and imaginary part of each subcarrier data. Of course, this method is very slow. I'm trying to replace outliers with something else.

*我还没有找到hampel滤波器处理复数的资料，对于csi数据，我分别用hampel滤波器处理每一个子载波的数据的实部和虚部，当然这种做法十分得慢。我正在尝试用别的办法替代异常值。*

closed 2023-04-17 by catcatBlue