espressif/esp-csi #104: 你好，我想问一下ESP32有没有双天线的？采集两份CSI信息

xiangshang-kok, 2023-04-13:
单天线在使用过程中，无法计算出相位差

MacChu0315-Espressif, 2023-04-13:
Sadly, nope, we have no such a module that can receive wifi signal by using two antennas at the same time by now.
One way to calculate the phase difference of two devices is to make them share the same crystal oscillator.

xiangshang-kok, 2023-04-14:
@MacChu0315-Espressif 
Thank you Sir, make them share the same crystal oscillator.Do you have any more specific operating methods?

MacChu0315-Espressif, 2023-04-18:
@xiangshang-kok Sorry, detailed info is not available. Relative modules will be realeased in the future.

zhanzhaocheng, 2023-05-31:
Hello, @xiangshang-kok 

Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng

kingaza, 2023-09-21:
> @xiangshang-kok Sorry, detailed info is not available. Relative modules will be realeased in the future.

@MacChu0315-Espressif when it will be released as plan?