espressif/esp-csi #106: How do I find the sampling rate for the extraction of esp32, where on menuconfig?

olliec28, 2023-04-18:


MacChu0315-Espressif, 2023-04-18:
What do you mean "sampling rate for the extraction of esp32", can you describe it in an other way?

olliec28, 2023-04-18:
> What do you mean "sampling rate for the extraction of esp32", can you describe it in an other way?

I basically want to know what the sampling rate is, is it 100Hz, 10Hz? How do i find this out?

MacChu0315-Espressif, 2023-04-18:
Therocally the sampling rate of ESP32 can reach millions per second.
The main bottleneck of sampling rate is sending rate, we used to sample CSI data at 3KHz.

MacChu0315-Espressif, 2023-04-18:
Oh, i believe that i get you now, there was a misguide in what i said.

> Therocally the sampling rate of ESP32 can reach millions per second. The main bottleneck of sampling rate is sending rate, we used to sample CSI data at 3KHz.

I mean, therocally the sampling rate of ESP32 is millions per second. How many CSI data you receive per second mainly depend on the sending rate and how many frames you lose when transmitting.

closed 2023-05-31 by zhanzhaocheng