espressif/esp-csi #108: Do you have plans for releasing source for `libesp-csi.a`?

jayavanth, 2023-04-21:


xiangshang-kok, 2023-04-21:
Sir,I would like to know what info content in the `libesp-csi.a` file

I am finding the `info->waveform_wander`parameter,how does it calculate，but noting found.Do you have any suggestion?

YunduanBeichen, 2023-04-22:
Similarly, in other issues, I found that the project can be run on esp32s2. But in actual operation, the libesp-csi.a file is missing in the lib folder. Could you please provide it?

MacChu0315-Espressif, 2023-05-10:
I am sorry that there are no plans to open source the libesp-csi.a for now.

MacChu0315-Espressif, 2023-05-10:
@YunduanBeichen Sorry no one got back to you earlier. If nothing important changed have been made, you can try to run the command `git checkout master` and `git reset --hard` to get the missing file.

zhanzhaocheng, 2023-05-31:
Hello, @jayavanth 

Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng

jayavanth, 2023-06-30:
Does `libesp-csi.a` also have Apache-2.0 as license? Can I use it commercially?

zhanzhaocheng, 2023-06-30:
@jayavanth libesp-csi.a is also Apache-2.0 and you can use it commercially