espressif/esp-csi #110: CSI receiver blocked

HKSIWAR, 2023-04-28:
Hi all,
I flashed two esp32 the first one is sender and the source code is in the following path :
https://github.com/espressif/esp-csi/tree/master/examples/get-started/csi_send
and the second is receiver and here is their path:
https://github.com/espressif/esp-csi/tree/master/examples/console_test
After 1 hour and 30 minutes, the sender works fine but the serial monitor of the receiver has been blocked, in fact the console does not display new data as shown in the following figure:
[image]

I have done this experience a few times and we always have the same problem.
What is the reason for this problem?

ahmed123-sys, 2023-04-28:
I have the same issue. Any suggestions!

MacChu0315-Espressif, 2023-05-05:
Thanks for discovering and reporting this problem. we have reproduced the problem and now is working on it.

zhanzhaocheng, 2023-06-07:
@ahmed123-sys This issue has been fixed, please update and verify it

closed 2023-06-12 by zhanzhaocheng