espressif/esp-csi #114: Understanding subcarrier selection in csi_data_read_parse.py

StrohmayerJ, 2023-05-23:
Hi, 

I am currently using the `get_started/tools/csi_data_read_parse.py` example for CSI data collection but I am a bit confused by the selection of valid subcarriers, which is performed as follows:

```
# Reduce displayed waveforms to avoid display freezes
CSI_VAID_SUBCARRIER_INTERVAL = 1

# Remove invalid subcarriers
# secondary channel : below, HT, 40 MHz, non STBC, v, HT-LFT: 0~63, -64~-1, 384
csi_vaid_subcarrier_index = []
csi_vaid_subcarrier_color = []
color_step = 255 // (28 // CSI_VAID_SUBCARRIER_INTERVAL + 1)

# LLTF: 52
csi_vaid_subcarrier_index += [i for i in range(6, 32, CSI_VAID_SUBCARRIER_INTERVAL)]     # 26  red
csi_vaid_subcarrier_color += [(i * color_step, 0, 0) for i in range(1,  26 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]
csi_vaid_subcarrier_index += [i for i in range(33, 59, CSI_VAID_SUBCARRIER_INTERVAL)]    # 26  green
csi_vaid_subcarrier_color += [(0, i * color_step, 0) for i in range(1,  26 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]
CSI_DATA_LLFT_COLUMNS = len(csi_vaid_subcarrier_index)

# HT-LFT: 56 + 56
csi_vaid_subcarrier_index += [i for i in range(66, 94, CSI_VAID_SUBCARRIER_INTERVAL)]    # 28  blue
csi_vaid_subcarrier_color += [(0, 0, i * color_step) for i in range(1,  28 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]
csi_vaid_subcarrier_index += [i for i in range(95, 123, CSI_VAID_SUBCARRIER_INTERVAL)]   # 28  White
csi_vaid_subcarrier_color += [(i * color_step, i * color_step, i * color_step) for i in range(1,  28 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]
# csi_vaid_subcarrier_index += [i for i in range(124, 162)]  # 28  White
# csi_vaid_subcarrier_index += [i for i in range(163, 191)]  # 28  White
```

When `CSI_VAID_SUBCARRIER_INTERVAL` is set to 1, I get 108 subcarriers (52 LLTF + 56 HT-LTF), most of which seem to give me usable data. 
[image]

My question is, how can we have >52 subcarriers if a single 802.11a channel only has 52 subcarriers? 
Also, in another [issue](https://github.com/espressif/esp-csi/issues/45) it was mentioned that valid subcarriers should be selected as follows:
```
# HT20 no STBC,   256 Byte
csi_data_vaid_index = []
# LLTF ==> 52
csi_data_vaid_index += [i for i in range(1,27)]      # 26
csi_data_vaid_index += [i for i in range(38, 64)]    # 26
# HT-LTF ==> 56
csi_data_vaid_index += [i for i in range(65, 93)]    # 28
csi_data_vaid_index += [i for i in range(100, 128)]  # 28

# HT40 no STBC,   384 Byte
select_list = []
# LLTF ==> 52
select_list += [i for i in range(6,32)]     # 26
select_list += [i for i in range(33,59)]    # 26
# HT-LTF  ==> 56 + 56
select_list += [i for i in range(66,94)]    # 28
select_list += [i for i in range(95,123)]   # 28
select_list += [i for i in range(134,162)]  # 28
select_list += [i for i in range(163,191)]  # 28
```
Why does `get_started/tools/csi_data_read_parse.py` use neither of these options but what seems to be HT40 no STBC with 256 Byte?
Finally, what is the highest number of valid subcarriers that I can use with the ESP32-S3?

I would appreciate it if someone could enlighten me. Thank you in advance!

zhanzhaocheng, 2023-05-26:
Your problem is detailed in the developer guide, see: [Wi-Fi Channel State Information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information)

StrohmayerJ, 2023-05-26:
@zhanzhaocheng Thank you for pointing me in the right direction. So if I understand correctly, according to these tables, the example uses HT40 non STBC, resulting in 108 usable subcarriers. As this configuration has no STBC-HT-LTF field, the number of received bytes is 256 instead of 384. Is my interpretation correct?

[image]

Also, according to the guide it is recommended to use HT20 in congested environments. 

"Theoretically, the HT40 can gain better throughput because the maximum raw physicial (PHY) data rate for HT40 is 150 Mbps while it is 72 Mbps for HT20. However, if the device is used in some special environment, e.g., there are too many other Wi-Fi devices around the ESP32-S3 device, the performance of HT40 may be degraded. **So if the applications need to support same or similar scenarios, it is recommended that the bandwidth is always configured to HT20.**"

How can I enable HT20 communication in the `get_started/tools/csi_data_read_parse.py` example? Is it enough to set `ESP_ERROR_CHECK(esp_wifi_set_bandwidth(ESP_IF_WIFI_STA, WIFI_BW_HT20));` for both transmitter and receiver?

Finally, the example uses a sending frequency of 100Hz set by `CONFIG_SEND_FREQUENCY`. Can this frequency be increased further, or is 100Hz the upper limit? When I try to increase it above 100Hz the transmitter throws errors and no packets are sent.  

Thank you!

zhanzhaocheng, 2023-05-30:
**If the current problem is solved, please close this issue, please submit a separate issue for each problem**

1.  **[Q]** Is it enough to set ESP_ERROR_CHECK(esp_wifi_set_bandwidth(ESP_IF_WIFI_STA, WIFI_BW_HT20)); for both transmitter and receiver?
**[A]**: Yes

2. **[Q]**: Can this frequency be increased further, or is 100Hz the upper limit?
**[A]**: You can use `usleep` instead of `vTaskDelay` to achieve a higher packet sending rate
    ```shell
    // vTaskDelay(pdMS_TO_TICKS(1000 / CONFIG_SEND_FREQUENCY));
    usleep(1000 * 1000 / CONFIG_SEND_FREQUENCY);
    ```

StrohmayerJ, 2023-05-31:
@zhanzhaocheng Solved, thank you!

closed 2023-05-31 by StrohmayerJ