espressif/esp-csi #115: Problem trying to use esp_wifi_radar_init() as there is undefined reference to `_esp_error_check_failed'

filzek, 2023-05-24:
C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\WiseHomeSwitch.elf.rsp -o WiseHomeSwitch.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/lib/libesp-csi.a(esp_radar.c.obj):(.literal.esp_wifi_radar_init+0x30): undefined reference to `_esp_error_check_failed'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/lib/libesp-csi.a(esp_radar.c.obj): in function `esp_wifi_radar_init':
/Users/zhanzhaocheng/project/esp-csi/esp-csi-gitlab/examples/console_test/components/esp-csi/src/esp_radar.c:659: undefined reference to `_esp_error_check_failed'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: /Users/zhanzhaocheng/project/esp-csi/esp-csi-gitlab/examples/console_test/components/esp-csi/src/esp_radar.c:660: undefined reference to `_esp_error_check_failed'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: /Users/zhanzhaocheng/project/esp-csi/esp-csi-gitlab/examples/console_test/components/esp-csi/src/esp_radar.c:672: undefined reference to `_esp_error_check_failed'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: /Users/zhanzhaocheng/project/esp-csi/esp-csi-gitlab/examples/console_test/components/esp-csi/src/esp_radar.c:673: undefined reference to `_esp_error_check_failed'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: ../main/lib/libesp-csi.a(esp_radar.c.obj):/Users/zhanzhaocheng/project/esp-csi/esp-csi-gitlab/examples/console_test/components/esp-csi/src/esp_radar.c:674: more undefined references to `_esp_error_check_failed' follow
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

We have used the unrelease libesp-csi.a with 16737KB it doesnt give this error, but also there is another problem, the use of local memory for the application, the malloc is trying to heap_caps_malloc was called but failed to allocate 32768 bytes with 0x804 capabilities, so, in the real usage we must target it to use spiram external memory to save local memory.

Please advise how to bypass those errors or provide a fix to it.

thanks.

zhanzhaocheng, 2023-05-26:
Please try again with a new repository, if you enable `CONFIG_SPIRAM_SUPPORT` it will be allocated in SPRAM by default in the latest libesp-csi.a
If the error still occurs, please provide the complete log file

filzek, 2023-05-29:
worked, so I will close this. But stil using 32k of local memory

closed 2023-05-29 by filzek