espressif/esp-csi #116: error new example with Rainmaker - iram overflow

filzek, 2023-05-29:
usind default example, just build and it give an error.

Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated D:/Dropbox/esp-csi/examples/connect_rainmaker/build/bootloader/bootloader.bin
[100/100] cmd.exe /C "cd /D D:\Dropbox\esp-csi\examples\connect_rainmaker\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:/Dropbox/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/Dropbox/esp-csi/examples/connect_rainmaker/build/bootloader/bootloader.bin"
Bootloader binary size 0x6b00 bytes. 0x500 bytes (4%) free.
[1242/1245] Linking CXX executable connect_rainmaker.elf
FAILED: connect_rainmaker.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\connect_rainmaker.elf.rsp -o connect_rainmaker.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: connect_rainmaker.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 5836 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

filzek, 2023-05-30:
Adjusted some extras from the default ane it works!

I will reset to default and so make a step by step to add here.

zhanzhaocheng, 2023-05-30:
Thank you very much for spotting this issue, what version of ESP-IDF are you using, I will fix this

closed 2023-05-30 by zhanzhaocheng