espressif/esp-csi #119: esp_radar.h not found or corrupt file

hasssaan007, 2023-06-02:
Thanks for this amazing git. I have got one issue, esp_radar.h file is not found by espressif ide although it is present in include folder. I also tried to place it along with app_main.c file but still same issue. Any help will means alot. I am pasting the error below:

Running ninja in directory 'c:\users\muhammad hassaan\onedrive\desktop\esp\examples\console_test\build'
Executing "ninja all"...
[1/10] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D "C:\Users\Muhammad Hassaan\OneDrive\Desktop\esp\examples\console_test\build\bootloader\esp-idf\esptool_py" && C:\Espressif2\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif2/frameworks/esp-idf-v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 "C:/Users/Muhammad Hassaan/OneDrive/Desktop/esp/examples/console_test/build/bootloader/bootloader.bin""
Bootloader binary size 0x69c0 bytes. 0x640 bytes (6%) free.
[2/8] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/radar_evaluate.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/radar_evaluate.c.obj
C:\Espressif2\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/newlib/platform_include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/freertos/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions/freertos -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc/esp32 -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/. -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/private_include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/heap/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/log/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/lwip/include/apps -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/lwip/include/apps/sntp -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/lwip/lwip/src/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include/arch -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/soc/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/soc/esp32/. -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/soc/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/hal/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/hal/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/hal/platform_port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_rom/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_rom/include/esp32 -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32 -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_common/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_system/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_system/port/soc -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_system/port/public_compat -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/xtensa/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/xtensa/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/driver/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/driver/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_pm/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_ringbuf/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/efuse/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/efuse/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/vfs/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_wifi/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_event/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_netif/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_eth/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/tcpip_adapter/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_phy/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_phy/esp32/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_ipc/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/app_trace/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_timer/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/mbedtls/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/mbedtls/mbedtls/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/mbedtls/esp_crt_bundle/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/app_update/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/spi_flash/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/bootloader_support/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/nvs_flash/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/pthread/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/xtensa -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/esp32 -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/espcoredump/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/espcoredump/include/port/xtensa -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/esp_supplicant/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/ieee802154/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/console -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/asio/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/cbor/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/unity/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/unity/unity/src -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/cmock/CMock/src -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/coap/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/coap/libcoap/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/nghttp/port/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/nghttp/nghttp2/lib/includes -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp-tls -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp-tls/esp-tls-crypto -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_adc_cal/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_hid/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/tcp_transport/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_http_client/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_http_server/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_https_ota/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_lcd/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_lcd/interface -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/protobuf-c/protobuf-c -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/protocomm/include/common -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/protocomm/include/security -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/protocomm/include/transports -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/mdns/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_local_ctrl/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/sdmmc/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_serial_slave_link/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/components/esp_websocket_client/include -IC:/Espressif2/frameworks/esp-idf-v4.4.1/ [... cut ...] -D_GNU_SOURCE -DIDF_VER=\"v4.4.1\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\app_main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj -c ../main/app_main.c
../main/app_main.c:32:10: fatal error: esp_radar.h: No such file or directory
 #include "esp_radar.h"
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

zhanzhaocheng, 2023-06-03:
You need to add the path of setting CSI in your Makefile, you can refer to: https://github.com/espressif/esp-csi/blob/master/examples/connect_rainmaker/CMakeLists.txt#L14
```shell
set(EXTRA_COMPONENT_DIRS ${CSI_PATH})
```

zhanzhaocheng, 2023-06-16:
Hello, @hasssaan007

Has the problem been resolved using the method me provided?

To prevent new issues from being buried among numerous existing ones, I have temporarily closed this issue. If your problem remains unresolved, please reopen this issue and provide additional details. Thanks.