espressif/esp-csi #121: Phase offsets in ESP32 CSI

a-arun1, 2023-06-14:
Hello, the phase measurements over packets are often corrupted by carrier frequency offsets, sampling frequency offsets or packet detection delays. 

An example is shown in the following image, where changing packet detection delays show up as varying slopes over multiple packets. More details are available in this [paper](https://ieeexplore.ieee.org/document/8423070). Having previously worked with the Nexmon and Intel CSI toolkit, I have observed these varying slopes. 

[image]

However, I do not observe this varying slope when we extract the CSI from the ESP 32 and plot the phases across subcarriers for different packets. See below, where each line is a different packet. The slopes are very consistent over packets.  

[image]

I'd like to understand if the reported CSI applied any corrections to remove these effects. 

Thank you for open-sourcing this tool, and I appreciate your thoughts on this!

zhanzhaocheng, 2023-06-15:
1. You can try the ESP32-S3 for CSI testing. The ESP32-S3 performs better than the ESP32.
2. Previously, we mainly focused on analyzing the amplitude of CSI. Regarding the phase that you mentioned, we will also replicate it first to pinpoint the specific issue. We will keep you updated promptly on any progress we make.

a-arun1, 2023-06-15:
We are currently using the ESP32-S3 module. This behavior has been observed previously by another user ( https://github.com/espressif/esp-csi/issues/101) as well. See Figure 3 replicated below:

[image]

You can see slope over the packets (Times axis) is very consistent. 

Do you have any documentation on how the CSI is determined?

zhanzhaocheng, 2023-06-15:
Thank you very much for your feedback. We will confirm the issue as soon as possible and expect to provide you with the results next week

julianstj1, 2023-06-15:
Hi Zhan, thanks for your quick reply on this.

I actually built a custom PCB with multiple ESP32-S3 chipsets on it targeting using CSI for this kind of phase detection for IOT prototyping. 

I thought it was something wrong in my software stack, but based on this post, it looks to be because of the behavior of the API. If there were more flags for the CSI API to enable or disable these kind of pre-processing the data before passing it up to the CSI application, this would be amazing and help us develop some very unique applications for ESP not currently possible with any other chipset.

Please let us know any update, thank you!

zhanzhaocheng, 2023-06-16:
**Q：If there were more flags for the CSI API to enable or disable these kind of pre-processing the data before passing it up to the CSI application**
**A**：Currently, it is possible to use [esp_wifi_set_csi_config()](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L1038-L1049) configuration, which API interfaces do you need

julianstj1, 2023-06-16:
Hi Zhan,

Based on the original comment here, it looks like there is collection of CSI after compensating for some kind of either carrier frequency offset or sampling frequency offset or one of the different asymmetries between the link partner and receiving device. As posted above, CSI should show properties of phase differences between different sub-carriers rather than a flat curve consistently.

So if it's possible, I think adding additional options to set_csi_config to disable any kind of carrier frequency offset or sampling frequency offset or whichever compensations are being used in the CSI calculation chain would be amazing.

Thank you!

a-arun1, 2023-06-16:
@zhanzhaocheng We have toggled all the configurations you have provided within CSI config, however, non of these turn off the slope corrections the original post mentioned.

MacChu0315-Espressif, 2023-06-21:
> We are currently using the ESP32-S3 module. This behavior has been observed previously by another user ( #101) as well. See Figure 3 replicated below:
> 
> [image]
> 
> You can see slope over the packets, in the right sub-figure (Times axis) is very consistent.
> 
> Do you have any documentation on how the CSI is determined?

@a-arun1 We have reproduced your issue and are consulting with our internal development team. We will keep you informed of any updates or progress regarding this matter.

MacChu0315-Espressif, 2023-06-26:
@a-arun1 After confirmation and discussion by our internal R&D team, our chip has no special processing process after receiving CSI signals. In the case of no path length change, the phase of each subcarrier of a channel should be consistent. According to your description, the phenomenon you reported may be due to the carrier frequency offset (CFO) and synchronization been processed. 
In the meantime, we are having internal discussions about the cause of some subcarrier spikes in amplitude and phase. We will keep you posted if there is any further news. If you have further questions or need further support, please do not hesitate to contact us.

a-arun1, 2023-06-27:
Hello @MacChu0315-Espressif, 

Thank you for testing and reproducing our issue. 

There may be a misunderstanding, as I do not completely agree with your comment. 

CFO is usually corrected before computing the channel state information. However, it is partially removed. Hence there is some residual CFO that remains in CSI. This residual CFO appears as a change in the intercept of the phases. This is clear from my first plot, where the different colored lines (corresponding to different packets over time) have changing intercepts. Residual CFO is not constant over packets; we are observing this from the phase plots. 

However, timing offsets, like time-of-travel, a consequence of the path length traveled by the signal, shows up as a slope over various channel subcarriers. The phase accumulated is ($2 \pi f_i \tau$ ), where $f_i$ are the subcarriers and $\tau$ is the time of travel. This would mean that there is a slope in the phase of the subcarriers and is _not_ consistent. 

But we are not observing this slope across the subcarriers, which leads me to believe there is pre-processing happening prior to reporting the CSI. Please feel free to let me know if you need additional details on this comment. 

There are additional details about this [here](https://www.mathworks.com/help/wlan/ug/joint-sampling-rate-and-carrier-frequency-offset-tracking.html)

Additionally, I would like to check if there's a possibility of exposing the CFO, which was estimated and corrected for before CSI estimation. I can open a new issue on this end if required as well.

MacChu0315-Espressif, 2023-06-28:
Hi, @a-arun1 . We will carefully review your message and provide a response as soon as possible. Thank you in advance for your patience.

a-arun1, 2023-07-11:
Hello @MacChu0315-Espressif, I wanted to follow up on my previous message. Please let me know if you'd like any additional information.

julianstj1, 2023-08-01:
Hello,
I'd also really like to hear from ESP about this property of the CSI library. I have my custom boards in hand now and this behavior is critical to understanding how the ESP chipsets can be used. 
Thank you!

MacChu0315-Espressif, 2023-08-01:
@julianstj1 @a-arun1 Sorry for the long delay in getting back to you and many thanks for being patient. I'm sorry to inform you that we still cannot provide a specific response at the moment. However, we can assure you that we have reviewed the paper you mentioned, conducted further in-depth internal discussions, and are currently designing additional experiments for verification. If there are any new results in the future, we will promptly notify you. Thank you again for your patience and understanding.

MacChu0315-Espressif, 2023-08-03:
@julianstj1 @a-arun1 Hi, we have made some progress and would like to keep you informed.
# Have done:
We performed the related experiment with a shielded box and two ESP32-C3-DevKitM-1. The testing circumstance is shown below.
[image]

# Data processing and analysis:
The figure below is the former 100 packets collected at the circumstance. Residual frequency offset and difference in initial phase can be clearly viewed.
[image]
Let's remove the offsets between packets by minousing the value of first subcarrier of each packet. The former 100 packet data after alignment is shown below.
[image]

# Conclusions:
The varying slope between packets can actually been observed on the CSI data collected by ESP chips, but is reletive stable. Such a phenomenon is due to the timing stability of ESP receiver.

# Add ones:
The pseudo code of removing the offsets is shown as follows: 
&ensp;&ensp;for i in subcarrier_index:
&ensp;&ensp;&ensp;&ensp;current_packet_angle.append(math.degrees(cmath.phase(current_subcarrier_complex * first_subcarrier_complex.conjugate())))

closed 2023-09-18 by zhanzhaocheng