espressif/esp-csi #123: I see output in the log but no waveform

jayavanth, 2023-06-26:
[image]

Using csi_recv_router example with ESP32-S3. If I check the output with `monitor`, I can see the CSI dump but can't see it in the python GUI

MacChu0315-Espressif, 2023-06-26:
The signal from esp32 is obtained by the host computer through the serial port, and the serial port is a single-send-single-receive transmission protocol, so you can only use the GUI or view the log in the existing solution.

jayavanth, 2023-06-26:
@MacChu0315-Espressif I didn't quite get it. Are you saying it can't read directly from the ESP and I have to upload the log?

Also, does the motion detection model work yet? Can you please point me to the code for that?

MacChu0315-Espressif, 2023-06-27:
I apologize for any confusion caused. If you wish to simultaneously open the serial port and obtain CSI dump while using the GUI, unfortunately, it is not possible. The GUI already opens the serial port to read the data, which includes CSI dump and algorithm results. The CSI dump is presented as a line chart within the GUI, rather than being printed in the GUI's LOG window.

If you would like to view the real-time raw CSI dump data while using the GUI, a simple approach is to modify the **csi_data_handle** function in the **esp_csi_tool.py** file to print the desired data.

Regarding the motion detection algorithm, it is functioning correctly. However, we are unable to disclose the specific implementation code at this time.

If you have any further inquiries or require additional assistance, please don't hesitate to let us know.

jayavanth, 2023-06-29:
Thanks @MacChu0315-Espressif. Is the motion detection happening in the ESP32 itself or in the python code?

zhanzhaocheng, 2023-06-29:
> Thanks @MacChu0315-Espressif. Is the motion detection happening in the ESP32 itself or in the python code?

The motion detection is implemented in ESP32, the python script simply displays the results of the detection

jayavanth, 2023-07-03:
@zhanzhaocheng I looked into the code a little bit. I see that the ESP32 only outputs CSI data.
`"type", "seq", "timestamp", "taget_seq", "taget", "mac", "rssi", "rate", "sig_mode", "mcs",
                          "cwb", "smoothing", "not_sounding", "aggregation", "stbc", "fec_coding","sgi", "noise_floor",
                          "ampdu_cnt", "channel_primary", "channel_secondary", "local_timestamp", "ant", "sig_len",
                          "rx_state", "len", "first_word_invalid", "data"`

 Then in the python code we can view the waveforms, record baselines for various movements like
`move, jump, sit down, etc.`

After these baselines are recorded, the GUI can tell you when those movements are detected by directly reading from the serial. Is that a correct summary of what is going on in the code?

zhanzhaocheng, 2023-07-07:
@jayavanth These are actions to label the raw data, which will be used as the data set for neural network training
Later we will also introduce how the neural network will be used to build models to identify different actions through CSI and deploy the models to ESP32.

jayavanth, 2023-07-07:
I see. That makes sense. Thank you!

closed 2023-07-07 by jayavanth