espressif/esp-csi #130: 仅显示两秒的波形

M3Captain7, 2023-08-12:
我使用esp32s3发送，使用esp32-s3-eye显示波形，第一次运行两个小时左右，esp-csi-tool显示波形都正常。接着重新烧录打开s3-eye的串口显示接收到信息，关闭串口使用esp-csi-tool仅显示两秒就停止了。
这个是串口收到的I (865) wifi:mode : null
I (865) wifi:Set ps type: 0

I (866) wifi:ic_enable_sniffer
I (867) esp_radar: CSI_RADAR Version: 0.1.0, Compile time: Jun  6 2023 - 14:20:50
I (868) esp_radar: csi_recv_interval: 10, csi_handle_time: 250, csi_handle_window: 50, csi_handle_buffer: 60
W (900) esp_radar: time_tamp: 589, end: 589, last: 0, num: 0
I (2405) app_main: ================ RADAR RECV ================
I (2406) app_main: type,sequence,timestamp,waveform_wander,someone_threshold,someone_status,waveform_jitter,move_threshold,move_status

RADAR_DADA,0,2407,0.000000,0.002000,0,0.000315,0.002000,0
RADAR_DADA,1,2645,0.000000,0.002000,0,0.000344,0.002000,0
RADAR_DADA,2,2896,0.000000,0.002000,0,0.000113,0.002000,0
RADAR_DADA,3,3136,0.000000,0.002000,0,0.000135,0.002000,0
RADAR_DADA,4,3396,0.000000,0.002000,0,0.000916,0.002000,0
RADAR_DADA,5,3636,0.000000,0.002000,0,0.000166,0.002000,0
RADAR_DADA,6,3900,0.000000,0.002000,0,0.000490,0.002000,0
RADAR_DADA,7,4150,0.000000,0.002000,0,0.000253,0.002000,0
RADAR_DADA,8,4384,0.000000,0.002000,0,0.000188,0.002000,0
RADAR_DADA,9,4646,0.000000,0.002000,0,0.000092,0.002000,0
RADAR_DADA,10,4896,0.000000,0.002000,0,0.000076,0.002000,0
仅显示两秒就暂停的波形
[image]

M3Captain7, 2023-08-12:
希望贵公司可以帮我解答，不胜感激

MacChu0315-Espressif, 2023-08-14:
Have you correctly disconnected the serial port of ESP32-S3-EYE? The phenomenon you've described closely resemble a situation where multiple application instances are accessing the serial port simultaneously. You can try disconnecting the ESP32-S3-EYE from your device and check if the issue persists. If this approach doesn't resolve your problem, please feel free to reach out to us at any time.

M3Captain7, 2023-08-14:
@MacChu0315-Espressif 谢谢你的回复。我使用idf.py monitor串口工具打开可以打印ESP32-S3-EYE的信息，接着使用 ctrl+] 断开串口工具，感觉是正确断开ESP32-S3-EYE的串口。如果没有正确断开串口直接使用CSI可视化工具，会提示如下错误：serial_handle: could not open port 'COM6': PermissionError(13, '拒绝访问。', None, 5)
Fial: Failed to open serial port。
现在依然是短暂打印就停止显示波形
[image]
直接使用串口可以正常一直输出
[image]

MacChu0315-Espressif, 2023-08-14:
Ok, we will attempt to reproduce your issue. If you could provide more detailed information, such as the commit hash of the esp-csi you are using and the IDF version number, it would be beneficial for resolving the problem.

M3Captain7, 2023-08-14:
@MacChu0315-Espressif 我使用的ESP-IDF 5.0.2 ,全程按照这个文档烧录例程https://github.com/espressif/esp-csi/blob/master/examples/esp-radar/console_test/README_cn.md，第一次正常运行两小时后我尝试使用ESP32-C3烧录csi_send，出现上述问题，接着换为ESP32S3问题依旧，不清楚和ESP32-C3是否有关系。请问在哪里可以看esp-csi 的 hash。

MacChu0315-Espressif, 2023-08-17:
Hi, @M3Captain7. 
In the `console_test` application, the default output method is using UART. However, the interface on the ESP32-S3-Eye board is USB. Therefore, you can try replacing the following code snippet in `console_test/app_main.c`:

```c
esp_console_repl_t *repl = NULL;
esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
repl_config.prompt = "csi>";
ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
```

With the following code:

```c
#if CONFIG_ESP_CONSOLE_UART
esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
#elif CONFIG_ESP_CONSOLE_USB_CDC
esp_console_dev_usb_cdc_config_t cdc_config = ESP_CONSOLE_DEV_CDC_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_console_new_repl_usb_cdc(&cdc_config, &repl_config, &repl));
#elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
esp_console_dev_usb_serial_jtag_config_t usbjtag_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&usbjtag_config, &repl_config, &repl));
#endif
```

Additionally, please modify the `channel for console output` option in the `menuconfig` from `Custom UART` to `USB Serial/JTAG Controller`.

Please let us know if this suggestion is effective. If you encounter any issues, please feel free to leave a message for further assistance.

M3Captain7, 2023-09-14:
谢谢你的回复，使用ESP32-S3-EYE可以正常显示波形，问题解决。
替换的内容需要加上这两行
esp_console_repl_t *repl = NULL;
esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
贵公司的专业与及时回复超出我预期，祝贵公司蒸蒸日上。

M3Captain7, 2023-09-14:
menuconfig  ->  component config  ->  ESP system setting -> channel for console output

closed 2023-09-14 by M3Captain7