espressif/esp-csi #131: how to differentiate between pet and human in detection ?

smtibaa, 2023-08-12:
pet is also detected using radar demo tool especially when it’s near sender or receiver and detection is always failing when there is a pet in sensing area 

there is any parameter to activate in order to exclude pet from human detection ?

Regards
Sabri

MacChu0315-Espressif, 2023-08-14:
We are sorry to inform you that the current version of the sample does not include this feature. However, we are actively developing new detection techniques and hope to make you experience our progress of our sample in the future.

smtibaa, 2023-08-14:
Thanks for your quick answer. Do you have a roadmap in particular detection technique?

MacChu0315-Espressif, 2023-08-17:
We apologize for any inconvenience, but we are currently unable to disclose specific plans. However, if you're interested in developing various applications using CSI, you might find this repository useful: [Awesome-WiFi-CSI-Sensing](https://github.com/Marsrocky/Awesome-WiFi-CSI-Sensing).

smtibaa, 2023-08-20:
Thanks a lot for sharing interesting documentation. One last question : do you use TinyML for human detection ? I know that it's not open source but I want just to have an idea about how to generate wander from collected CSI ?

closed 2023-09-18 by zhanzhaocheng