espressif/esp-csi #132: Board flashed but no data from serial

pablo-rebolini-treinta, 2023-09-10:
Hello there, i have a ESP-WROOM-32 board and compiled esp-radar is done without major problems.
The thing is, when i connect the board to my mac and open the gui tool, no data is visible in the logs box, (also tried pressing EN button). But if i connect to the board using a serial client, i see some information, next will share with you technical details, i will be so gratefully if someone can help me.

Board: ESP-WROOM-32
Chip: esp32 (i tried with esp32s2 but i got "wrong chip message from esptool)
Baud: 115200

This is the only data i get from the board:

```
Executing action: flash
Running ninja in directory /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build
Executing "ninja flash"...
[1/5] cd /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build/esp-idf/esptool_py && /U...tition-table.bin /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build/console_test.bin
console_test.bin binary size 0xd57e0 bytes. Smallest app partition is 0x1ca000 bytes. 0xf4820 bytes (53%) free.
[1/1] cd /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build/bootloader/esp-idf/espto... 0x1000 /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build/bootloader/bootloader.bin
Bootloader binary size 0x7000 bytes. 0x0 bytes (0%) free.
[4/5] cd /Users/blackbox/esp/esp-idf/components/esptool_py && /opt/homebrew/Cellar/cmake/3.27.4/bin...radar/console_test/build -P /Users/blackbox/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32 -p /dev/cu.usbserial-56860524611 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x20000 console_test.bin 0x8000 partition_table/partition-table.bin 0x1d000 ota_data_initial.bin
esptool.py v4.7.dev1
Serial port /dev/cu.usbserial-56860524611
Connecting.....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:a7:32:de:f3:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00020000 to 0x000f5fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0001d000 to 0x0001efff...
Compressed 28672 bytes to 17925...
Writing at 0x0000784d... (100 %)
Wrote 28672 bytes (17925 compressed) at 0x00001000 in 1.9 seconds (effective 122.7 kbit/s)...
Hash of data verified.
Compressed 874464 bytes to 587719...
Writing at 0x000f0c0a... (100 %)
Wrote 874464 bytes (587719 compressed) at 0x00020000 in 56.8 seconds (effective 123.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 148...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (148 compressed) at 0x00008000 in 0.0 seconds (effective 740.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0001d000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0001d000 in 0.0 seconds (effective 1856.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test
Executing "/Users/blackbox/.espressif/python_env/idf5.2_py3.11_env/bin/python /Users/blackbox/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-56860524611 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/build/console_test.elf -m '/Users/blackbox/.espressif/python_env/idf5.2_py3.11_env/bin/python' '/Users/blackbox/esp/esp-idf/tools/idf.py' '-b' '115200' '-p' '/dev/cu.usbserial-56860524611'"...
--- idf_monitor on /dev/cu.usbserial-56860524611 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7488
load:0x40078000,len:16772
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4304
entry 0x40080670
I (31) boot: ESP-IDF v5.2-dev-2708-g3640dc86bb-dirty 2nd stage bootloader
I (31) boot: compile time Sep  9 2023 22:11:48
I (33) boot: Multicore bootloader
I (38) boot: chip revision: v3.0
I (41) qio_mode: Enabling default flash chip QIO
I (47) boot.esp32: SPI Speed      : 80MHz
I (51) boot.esp32: SPI Mode       : QIO
I (56) boot.esp32: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (84) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (92) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (99) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (107) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (114) boot:  5 ota_0            OTA app          00 10 00020000 001ca000
I (122) boot: End of partition table
I (126) boot: No factory image, trying OTA 0
I (131) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=257cch (153548) map
I (182) esp_image: segment 1: paddr=000457f4 vaddr=3ffb0000 size=036c0h ( 14016) load
I (186) esp_image: segment 2: paddr=00048ebc vaddr=40080000 size=0715ch ( 29020) load
I (197) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=962cch (615116) map
I (366) esp_image: segment 4: paddr=000e62f4 vaddr=4008715c size=0f4c4h ( 62660) load
I (398) boot: Loaded app from partition at offset 0x20000
I (404) boot: Set actual ota_seq=1 in otadata[0]
I (404) boot: Disabling RNG early entropy source...
I (415) cpu_start: Multicore app
I (415) cpu_start: Pro cpu up.
I (415) cpu_start: Starting app cpu, entry point is 0x4008132c
0x4008132c: call_start_cpu1 at /Users/blackbox/esp/esp-idf/components/esp_system/port/cpu_start.c:178

I (0) cpu_start: App cpu up.
I (433) cpu_start: Pro cpu start user code
I (433) cpu_start: cpu freq: 240000000 Hz
I (433) cpu_start: Application information:
I (438) cpu_start: Project name:     console_test
I (444) cpu_start: App version:      6b8b923-dirty
I (449) cpu_start: Compile time:     Sep  9 2023 22:11:40
I (455) cpu_start: ELF file SHA256:  be35126d4...
I (461) cpu_start: ESP-IDF:          v5.2-dev-2708-g3640dc86bb-dirty
I (468) cpu_start: Min chip rev:     v0.0
I (472) cpu_start: Max chip rev:     v3.99 
I (477) cpu_start: Chip rev:         v3.0
I (482) heap_init: Initializing. RAM available for dynamic allocation:
I (489) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (495) heap_init: At 3FFB8080 len 00027F80 (159 KiB): DRAM
I (501) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (514) heap_init: At 40096620 len 000099E0 (38 KiB): IRAM
I (521) spi_flash: detected chip: generic
I (525) spi_flash: flash io: qio
I (529) app_start: Starting scheduler on CPU0
I (534) app_start: Starting scheduler on CPU1
I (534) main_task: Started on CPU0
I (544) main_task: Calling app_main()
W (547) ws2812_led: WS2812 LED is disabled
```

Thanks in advance

pablo-rebolini-treinta, 2023-09-10:
UPDATE:
I did some changes in sdkconfig file and tryied again, got some traffic in idf monitor mode, but i think could be a baud rate misconfiguration? 

```
I (29) boot: compile time Sep 10 2023 15:22:34
I (34) boot: Multicore bootloader
I (42) boot: chip revision: v3.0
I (50) qio_mode: Enabling default flash chip QIO
I (60) boot.esp32: SPI Speed      : 40MHz
I (70) boot.esp32: SPI Mode       : QIO
I (79) boot.esp32: SPI Flash Size : 4MB
I (88) boot: Enabling RNG early entropy source...
I (99) boot: Partition Table:
I (106) boot: ## Label            Usage          Type ST Offset   Length
I (121) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (136) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (151) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (166) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (181) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (196) boot:  5 ota_0            OTA app          00 10 00020000 001ca000
I (211) boot: End of partition table
I (220) boot: No factory image, trying OTA 0
I (230) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=2880ch (165900) map
I (272) esp_image: segment 1: paddr=00048834 vaddr=3ffb0000 size=036c0h ( 14016) load
I (275) esp_image: segment 2: paddr=0004befc vaddr=40080000 size=0411ch ( 16668) load
I (288) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=98778h (624504) map
I (397) esp_image: segment 4: paddr=000e87a0 vaddr=4008411c size=124d4h ( 74964) load
I (424) boot: Loaded app from partition at offset 0x20000
I (428) boot: Set actual ota_seq=1 in otadata[0]
I (428) boot: Disabling RNG early entropy source...
I (458) cpu_start: Multicore app
I (458) cpu_start: Pro cpu up.
I (458) cpu_start: Starting app cpu, entry point is 0x4008132c
0x4008132c: call_start_cpu1 at /Users/blackbox/esp/esp-idf/components/esp_system/port/cpu_start.c:178

I (0) cpu_start: App cpu up.
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 240000000 Hz
I (494) cpu_start: Application information:
I (504) cpu_start: Project name:     console_test
I (515) cpu_start: App version:      6b8b923-dirty
I (526) cpu_start: Compile time:     Sep 10 2023 15:22:27
I (538) cpu_start: ELF file SHA256:  1c9a6ef82...
I (549) cpu_start: ESP-IDF:          v5.2-dev-2708-g3640dc86bb-dirty
I (563) cpu_start: Min chip rev:     v0.0
I (572) cpu_start: Max chip rev:     v3.99 
I (582) cpu_start: Chip rev:         v3.0
I (591) heap_init: Initializing. RAM available for dynamic allocation:
I (606) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (618) heap_init: At 3FFB8108 len 00027EF8 (159 KiB): DRAM
I (630) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (643) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (656) heap_init: At 400965F0 len 00009A10 (38 KiB): IRAM
I (669) spi_flash: detected chip: generic
I (677) spi_flash: flash io: qio
I (686) app_start: Starting scheduler on CPU0
I (695) app_start: Starting scheduler on CPU1
I (695) main_task: Started on CPU0
I (714) main_task: Calling app_main()
W (722) ws2812_led: WS2812 LED is disabled
```

Monitor output:

[image]

Thanks

MacChu0315-Espressif, 2023-09-11:
Thanks for the detailed bug report and letting us know about this. 
We reproduced the problem you reported and found out that the cause of it is the API change of setting baudrate. You can add `#include "driver/uart.h"` and replace `uart_ll_set_baudrate` with `uart_set_baudrate(UART_LL_GET_HW(CONFIG_ESP_CONSOLE_UART_NUM), CONFIG_ESP_CONSOLE_UART_BAUDRATE);` in file `app_main.c`. 
Please let us know whether the method above helps and feel free to reach us for further assistance.

pablo-rebolini-treinta, 2023-09-13:
Hey @MacChu0315-Espressif i really appreciate you response, thanks for your time.
I applied the changes you suggested and the results was this:

```
/Users/blackbox/esp/esp-idf/components/hal/esp32/include/hal/uart_ll.h:27:63: error: incompatible type for argument 1 of 'uart_set_baudrate'
   27 | #define UART_LL_GET_HW(num) (((num) == UART_NUM_0) ? (&UART0) : (((num) == UART_NUM_1) ? (&UART1) : (&UART2)))
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                               |
      |                                                               volatile uart_dev_t * {aka volatile struct uart_dev_s *}
/Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/main/app_main.c:522:31: note: in expansion of macro 'UART_LL_GET_HW'
  522 |             uart_set_baudrate(UART_LL_GET_HW(CONFIG_ESP_CONSOLE_UART_NUM), CONFIG_ESP_CONSOLE_UART_BAUDRATE);
      |                               ^~~~~~~~~~~~~~
In file included from /Users/blackbox/esp/esp-csi/examples/esp-radar/console_test/main/app_main.c:38:
/Users/blackbox/esp/esp-idf/components/driver/uart/include/driver/uart.h:235:41: note: expected 'uart_port_t' but argument is of type 'volatile uart_dev_t *' {aka 'volatile struct uart_dev_s *'}
  235 | esp_err_t uart_set_baudrate(uart_port_t uart_num, uint32_t baudrate);
```

Some weird type mismatch error that i cant decipher. But... Reading the code a little i try this:
```
uart_set_baudrate(0, 115200);
```

And it was magic, its works, now i have a full ui working... Thanks again

I dont want to abuse of your help, but, do you have some insight of why the ui is not saving any data in data/ folder?

Thanks again!

pablo-rebolini-treinta, 2023-09-13:
Nevermind, storing data is working perfectly. Thanks again, will close the issue

closed 2023-09-13 by pablo-rebolini-treinta