espressif/esp-csi #135: can esp-csi be used in realtime case?

kingaza, 2023-09-14:
and what is the minimal delay?

MacChu0315-Espressif, 2023-09-14:
We are not entirely clear about your question. However, we can inform you that it takes approximately 50 nanoseconds from the moment the device receives a wireless signal packet to entering the CSI callback function. Please feel free to reach us for further assistance.

kingaza, 2023-09-14:
> We are not entirely clear about your question. However, we can inform you that it takes approximately 50 nanoseconds from the moment the device receives a wireless signal packet to entering the CSI callback function. Please feel free to reach us for further assistance.

thx for your reply.
For example, I want to recognize the human behavior as quickly as possible. therefore, I want to achieve the minimal delay time.
It is declared that some WiFi7 chips has a delay time about 2 or 3 ms.

MacChu0315-Espressif, 2023-09-14:
if we understand correctly, you're inquiring about the maximum receiving frequency that the ESP32 chip can support?
We used to conduct tests and have achieved a receiving frequency of approximately 3000Hz.

kingaza, 2023-09-14:
It sounds great! So, does it mean, that I can get the information just within 0.3ms?
btw, could you share the test cases and detailed results, or show me how to do the tests?
thanks a lot~

MacChu0315-Espressif, 2023-09-18:
For sure, we can provide some tips on reproducing the testing case:
1. Receiving freq depends heavily on the CSI sending freq, so you should try your best to boost it. To achieve that, you can reduce the sending interval and writes to terminals (your intended writes and warnings from IDF).
2. Similarly, you should write to terminals only when necessary for the receiver and _ets_printf_ is recommended in such a case. 
4. A recommend data saving manner is to store your CSI data into TF cards with SDIO protocol (since you are using S3) and don't put your TF card writing codes in the csi callback but a task instead.
5. To reduce the time cost on data saving, try to encode the data with, for example, base64 before writing to the card.

Add ones:
Transmission at too high a frequency is a pollution to the channel, remember to evaluate the impact of CSI transmission on other applications

kingaza, 2023-09-18:
> For sure, we can provide some tips on reproducing the testing case:
> 
> 1. Receiving freq depends heavily on the CSI sending freq, so you should try your best to boost it. To achieve that, you can reduce the sending interval and writes to terminals (your intended writes and warnings from IDF).
> 2. Similarly, you should write to terminals only when necessary for the receiver and _ets_printf_ is recommended in such a case.
> 3. A recommend data saving manner is to store your CSI data into TF cards with SDIO protocol (since you are using S3) and don't put your TF card writing codes in the csi callback but a task instead.
> 4. To reduce the time cost on data saving, try to encode the data with, for example, base64 before writing to the card.
> 
> Add ones: Transmission at too high a frequency is a pollution to the channel, remember to evaluate the impact of CSI transmission on other applications.

wow, so detailed! thank you very much!
So, it is enough in coding part that I only need to modify the following value to a smaller number, say 1, in the example get-started:
`#define CONFIG_SEND_FREQUENCY               100`

Actually I have tried this before, but unfortunately, I didn't know how to show or save data in such a fast way.

MacChu0315-Espressif, 2023-09-18:
No, on the contrary, you should increase the value of `CONFIG_SEND_FREQUENCY` because the size of this value is inversely proportional to the sending interval.  If necessary, you can comment out the sending wait function.

Regarding the issues with displaying and saving data, once you've confirmed that there is no overlap with any previously resolved issues, please feel free to submit detailed materials and discriptions in a new issue.

kingaza, 2023-09-26:
请问如果要测试3000Hz的接收频率，你们用的是哪一款开发板呢？淘宝店铺中这个没有SD卡槽
https://item.taobao.com/item.htm?spm=a1z10.5-c.w4002-8715811636.11.3c4d67d83D84gK&id=653155344338

我手里倒是还有个Box-lite，但不知道PCB天线是否影响？

MacChu0315-Espressif, 2023-09-26:
We perform the experiment with a ESP32-C3-DevKitM-1. To store the CSI data, we added a TFcard read / write module. The Box-lite share the same module (PCB antenna) with the S3 devkit, so there should not be much difference.

kingaza, 2023-09-26:
Thx.
I hope it doesn't bother you too much... could you share the log from your experiment? 
thx again~

MacChu0315-Espressif, 2023-09-26:
sorry that we are not ready to share the raw data. However, we'd love to share a pic of part of the data.
[image]
Note that the column D stands for the timestamp esp receives the packet, and the receive freq of this data is about 2KHz.

kingaza, 2023-09-26:
> sorry that we are not ready to share the raw data. However, we'd love to share a pic of part of the data. [image] Note that the column D stands for the timestamp esp receives the packet, and the receive freq of this data is about 2KHz.

OK, I would like to make it clear. So, I guess the timestamp format is "%S%f".
for example, the first timestamp shall be 15.397208 s.
therefore, the frequency = 1 / ((15.410685 - 15.397208) / 24) = 1780

MacChu0315-Espressif, 2023-09-26:
yes, that's right

kingaza, 2023-09-27:
Concerning about data saving, I have no idea about how to use a task (xTaskCreate?) yet, after reading the sdio examples ...
could you show me some code snippet? 
Sorry for troubling you again...

> 3. A recommend data saving manner is to store your CSI data into TF cards with SDIO protocol (since you are using S3) and don't put your TF card writing codes in the csi callback but a task instead.

MacChu0315-Espressif, 2023-10-07:
I'm sorry, but we don't have any plans to release the code related to this.
The usage of the SDIO example depends on an TF card read/write module. To create a task for data writing, you will need to modify the program yourself.

kingaza, 2023-11-03:
tested send frequency.
thx

closed 2023-11-03 by kingaza