espressif/esp-csi #136: esp csi是不是不能和esp now共用？

inxy-n, 2023-09-16:
尊敬的乐鑫工程师，您好：
我想只使用一块esp32-s3和路由器把csi检测结果通过无线信号发出去，经过实践，如果直接通过tcp向路由器发数据会严重干扰csi的检测结果。因此我尝试使用esp now协议，但是经过半个月的代码调整，最后输出的错误是ESP_ERR_ESPNOW_IF ，Interface error。esp now的信道已经调整为连接到的wifi的信道。至此找不到进一步的解决方案了。
因此是否只能采用再载一块esp8266之类的芯片或者在需要发esp now包的时候临时关闭csi然后再发包的方法呢？
不胜感谢。

inxy-n, 2023-09-16:
Does esp csi not work with esp now?

 
Dear espressif engineer,

I want to use only an esp32-s3 and router to send the csi detection results through wireless signals. After practice, sending data directly to the router through tcp will seriously interfere with the csi detection results. So I tried to use esp-now protocol, but after half a month of code adjustment, the output error was "ESP_ERR_ESPNOW_IF", "Interface error". The esp-now channel has been adjusted to the channel connected to the wifi. No further solution can be found.

So is it only possible to load another chip such as the esp8266 or temporarily turn off the csi when the esp-now package needs to be sent?

Thank you very much.

MacChu0315-Espressif, 2023-09-18:
ESP-CSI is able to coexist with esp-now.

To achieve that based on our repo, you can attach the esp-now sending code below ***wifi_ping_router_start();*** in _csi_recv_router/main/app_main.c_. And the esp-now sending code can be found below ***wifi_init();*** in _csi_send/main/app_main.c_ (don't forget to attach the ***CONFIG_LESS_INTERFERENCE_CHANNEL*** also) .

Please feel free to reach us for further assistance.

inxy-n, 2023-09-30:
@MacChu0315-Espressif. I'm very appreciated to you rapid reply and I'm sorry that I can't reply in time. I have flashed the programme you mentioned to my esp32, and I try to receive the esp-now message in my esp8266,but it received nothing. Is there some problem about channel? I see the sender's channel is set to 11,but I don't know how to change the receiver's channel in arduino IDE.Thanks again for your reply!
[image]
[image]

MacChu0315-Espressif, 2023-10-07:
Based on the image you've provided, it appears that your sender device successfully sent out the wireless packet. However, you mentioned not receiving the payload in ESP-NOW. Could you please provide the code for the receiver?