espressif/esp-csi #138: CSI的幅值不稳定，跳动剧烈

iSuryxin, 2023-09-23:
你好，在我按照esp-csi-master\examples\esp-radar\console_test里的操作文档对esp32-s3进行接收 Wi-Fi CSI程序烧录后，与一台商用路由器进行了连接，在顺利获取csi数据后，我将空房间的csi幅值进行绘制，我认为这种状态的幅值应该是比较平稳的，但我发现csi幅值很不稳定，一段时间是相对稳定，另一段时间又向两端跳跃，为什么会这样，还是说这是正常的，如果不是，可以给我一点意见吗，这对我十分重要，谢谢了。下图是时长为20分钟的第一条子载波幅值数据
[image]
[image]

MacChu0315-Espressif, 2023-09-25:
Hello,

May I inquire about the model of the commercial router you are using? How many antennas are there on it, and how many of them are operational during your experiments?

The reason we asked this because the router's antenna switching strategy is not always constant. If there was any antenna switching happening during your experiments, it could lead to fluctuations in the received CSI data.

iSuryxin, 2023-09-25:
> Hello,
>
> May I inquire about the model of the commercial router you are using? How many antennas are there on it, and how many of them are operational during your experiments?
>
> The reason we asked [...]
非常感谢，我使用的路由器的型号是“AC1200双频无线路由器 TL-WDR5620千兆易展版”，它有2根外置2.4GHz不可拆卸全向天线加2根外置5GHz不可拆卸全向天线，出现上述这种情况我还在想是否可以通过修改esp32的配置菜单，看看是否可以获得更稳定的幅值。但也尚不清楚

MacChu0315-Espressif, 2023-09-26:
We believe that modifying menuconfig will not help. One solution is to try another router, and the other solution is to manully set the ping rate to 11b (with esp_wifi_set_protocol function).

iSuryxin, 2023-09-26:
> 我们认为修改菜单配置将无济于事。一种解决方案是尝试另一个路由器，另一种解决方案是将ping速率设置为11b（具有esp_wifi_set_protocol功能）。

感谢你的指导，我还有一些不解的问题，第二种解决方案是指在你们esp-csi-master\examples\esp-radar\console_tes中的app_main.c的代码中添加ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_WIFI_PROTO_11B));非常感谢！

MacChu0315-Espressif, 2023-09-27:
Yes, please feel free to reach us for further assistance.

iSuryxin, 2023-09-27:
> Yes, please feel free to reach us for further assistance.
非常感谢您，我换了一台路由器，发现数据确定更加稳定多了，但是另一种解决方案在代码中添加ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_WIFI_PROTO_11B));，重新烧录完成后，在连接路由器的时候，log窗口出现spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k) 这样的错误提示，无法成功连接路由器。

MacChu0315-Espressif, 2023-10-07:
Initially, because we weren't sure whether the fluctuation in CSI amplitude was due to router antenna switching or rate negotiation switching, we proposed two approaches to test these hypotheses separately. If changing the router seems to work, it suggests that the issue is related to the router switching antennas. If the latter approach works, it implies that the instability in amplitude occurs during rate negotiation and switching. Based on your response, it seems that the cause of the problem is the former.

Regarding the issue of not being able to connect to the router, it appears that the spi_flash message is more of a warning than an error, so your program should be able to flash and run correctly, right? This doesn't seem to directly lead to an inability to connect to the router.

However, since we have limited information, could you please try the following command and let us know the result:

```c
ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_WIFI_PROTO_11B|ESP_WIFI_PROTO_11G|ESP_WIFI_PROTO_11N));
```

We allow ESP devices to choose and switch between the three protocols (BGN) without concerns about causing amplitude variations for the following reasons: 1. If the network environment is stable, rate switching occurrences are not frequent. 2. For extracting CSI data, these three protocols share the same LLTF.

Please provide us with the results, and we will continue to assist you further.

iSuryxin, 2023-10-10:
是的，你是对的，在你前面提到的将ping速率设置为11b，于是我设想将esp和路由器均设置为b协议，这将会导致esp无法连接其他的协议，非常感谢你！

MacChu0315-Espressif, 2023-10-11:
> 是的，你是对的，在你前面提到的将ping速率设置为11b，于是我设想将esp和路由器均设置为b协议，这将会导致esp无法连接其他的协议，非常感谢你！

I'm sorry, it appears there was some confusion in our previous response. To obtain CSI data, it's essential to ensure that the airborne packets adhere to protocols such as 802.11 a/g/n/ac/ax, rather than the 802.11b protocol. We mistakenly confused 802.11a with 802.11b. The reason the 11b protocol cannot provide CSI data is due to the absence of the corresponding structures in its packet header. Therefore, even if you successfully send and receive 11b packets, you won't be able to extract CSI data from them.

Yanc14, 2023-10-12:
> I'm sorry, it appears there was some confusion in our previous response. To obtain CSI data, it's essential to ensure that the airborne packets adhere to protocols such as 802.11 a/g/n/ac/ax, rather [...]

能否获得CSI信息为什么会与wifi协议版本相关？我认为能否获得CSI信息应该与芯片相关。

MacChu0315-Espressif, 2023-10-16:
@Yanc14 For in various WiFi protocols, the MAC frame header structure used to provide CSI data varies.

Yanc14, 2023-10-21:
> @Yanc14 For in various WiFi protocols, the MAC frame header structure used to provide CSI data varies.

thanks for your answer