espressif/esp-csi #143: 请问如何解决这个问题？

Yanc14, 2023-10-13:
[image]
[image]

Yanc14, 2023-10-13:
> [image] [image]

板卡是esp32，没有rgb-led，于是更改了gpio为gpio2

MacChu0315-Espressif, 2023-10-16:
I'm sorry, but we didn't quite understand your question. Could you please provide more details to help us better comprehend your issue?

cafercicek, 2023-10-21:
Hi, I have same issue. I flashed the console_test binary. I saw same result on the monitor. When I tried app GUI,  I didn't get some signal.

Yanc14, 2023-10-21:
> I'm sorry, but we didn't quite understand your question. Could you please provide more details to help us better comprehend your issue?

我查看了console_test下的app_main.c文件
void app_main(void)
{
    /**
     * @brief Install ws2812 driver, Used to display the status of the device
     */
    ws2812_led_init();
    ws2812_led_clear();
.........
}
这里需要对install ws2812 driver，我使用的是ESP32-DevKitC开发板，没有RGB灯，因此会产生如上问题，即不会有csi数据产生

MacChu0315-Espressif, 2023-10-22:
Hi, @Yanc14 and @cafercicek, since this LED bead serves solely for visual feedback and doesn't pertain to the core functionality, you can simply comment out all the sections of code related to ws2812 in the program.

Yanc14, 2023-10-23:
> Hi, @Yanc14 and @cafercicek, since this LED bead serves solely for visual feedback and doesn't pertain to the core functionality, you can simply comment out all the sections of code related to ws2812 in the program.

i did it , but didn't works.
[image]

Yanc14, 2023-10-23:
> Hi, @Yanc14 and @cafercicek, since this LED bead serves solely for visual feedback and doesn't pertain to the core functionality, you can simply comment out all the sections of code related to ws2812 in the program.
我注释掉ws2812的代码之后，build，flash and  python esp_csi_tool.py -p COM7，单数出现如下问题
[image]

Dafeigy, 2024-01-16:
I encountered the same problem. Have you solved it? On monitor there's a `I (790) main_task: Returned from app_main()` info and I think that's why I failed