espressif/esp-csi #145: csi子载波索引

Lijianping77, 2023-10-19:
请问esp32s3用示例代码examples/get-started/csi_recv_router
CSI_DATA,774,a2:6b:f8:4a:4a:87,-20,11,1,7,0,1,1,0,1,0,1,-97,0,1,0,13230366,0,83,0,128,0,"
[0,0,2,5,5,5,7,8,9,11,10,10,11,11,12,11,11,10,12,12,11,12,10,10,8,6,6,4,5,2,3,0,1,-3,0,-6,-1,-9,-1,-12,-2,-15,-2,-17,-2,-15,-2,-16,-1,-16,1,-18,2,-16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,15,-8,16,-8,13,-7,13,-6,12,-5,14,-3,11,-2,9,-2,5,-1,1,-2,-2,-2,-5,-3,-8,-4,-10,-4,-12,-5,-15,-7,-16,-8,-17,-8,-15,-6,-15,-6,-14,-5,-13,-5,-12,-5,-9,-2,-6,-1,-4,0]"
其中有多少条子载波？
或者esp32采集的子载波数量是可以自己改变的吗？

MacChu0315-Espressif, 2023-10-20:
CSI raw data starts from "[" and ends with "]", as each subcarrier occupies two bytes of data. In total, there are 64 subcarriers' worth of data here. The specific subcarriers that the esp32 can capture vary depending on the experimental configuration. You can refer to the [Wi-Fi Channel State Information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information)  documentation for details. For instance, when using "second channel + HT + 40MHz bandwidth + non STBC," the esp32 can receive a maximum of 384 bytes of data, which translates to data from 192 subcarriers.

If you wish to collect data from only specific subcarriers, you can achieve this by modifying the csi_config within the recv program.

Lijianping77, 2023-10-20:
> CSI原始数据从“[”开始，以“]”结束，因为每个子载波占用两个字节的数据。这里总共有64个子运营商的数据。esp32 可以捕获的特定子载波因实验配置而异.有关详细信息，请参阅 [Wi-Fi 信道状态信息](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information)文档。例如，当使用“第二通道 + HT + 40MHz 带宽 + 非 STBC”时，esp32 最多可以接收 384 字节的数据，这相当于来自 192 个子载波的数据.
> 
> 如果您只想从特定的子载波收集数据，您可以通过修改 recv 程序中的csi_config来实现这一点。

[image]
那么这64个子载波怎么变成复数形式呢？他们之间的对应关系是怎样的，还有如何与您文档对应呢？
另外您examples/esp-radar/console_test/tools radar data
printf("RADAR_DADA!,%d,%s,%.6f,%.6f,%d,%.6f,%.6f,%d\n",
           s_count++, timestamp_str,
           info->waveform_wander, g_console_input_config.predict_someone_threshold, room_status,
           info->waveform_jitter, g_console_input_config.predict_move_threshold, human_status);
是从CSI_DATA得到的吗？我在您代码中没找到，非常感谢您！！！！

Yanc14, 2023-10-21:
> > CSI原始数据从“[”开始，以“]”结束，因为每个子运营商占用了两个字节的数据。这里总共有64个子运营商的数据。esp32可以抓取特定的子运营商因实验配置而异。有关详细信息，请参阅[Wi-Fi 信道状态信息](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information)文档。例如，当使用“第二通道 + HT + 40MHz 带宽 + 非 STBC”时，esp32 最多接收 384 字节的数据，可以连接到来自 192 个子载波的数据。
> > 如果您从特定的子载波收集数据，您可以通过修改recv程序中的csi_config来实现这一点。
> 
> [image] 现在这64个子承载怎么变成复数形式呢？他们之间的对应关系是怎样的，还有如何与您的文档对应呢？另外您的examples/ esp-radar/console_test/tools 雷达数据 printf("RADAR_DADA!, %d、%s、%.6f、%.6f、%d、%.6f、%.6f、%d\n"、s_count ++、timestamp_str、 信息->waveform_wander、g_console_input_config.predict_someone_threshold、room_status、 信息->waveform_jitter , g_console_input_config.predict_move_threshold, human_status); 是从CSI_DATA得到的吗？我在您的代码中没有找到，非常感谢您！！！！

感觉每个子载波的实部占据8bit，虚部占据8bit，但是怎么转换为float就不知道了

MacChu0315-Espressif, 2023-10-22:
Hi @Lijianping77,

Regarding CSI raw data, every 2 bytes of data correspond to one subcarrier, with the first byte representing the imaginary part and the second byte representing the real part. Let's take the scenario of "second channel (above) + HT + 40MHz bandwidth + non STBC" as an example. The first 128 bits are derived from the LLTF (corresponding to subcarrier indices in the channel ranging from -64 to -1). These 128 bytes of data represent the imaginary part of the -64 subcarrier, the real part of the -64 subcarrier, the imaginary part of the -63 subcarrier, and so on.

As for your second question, these data are provided by non-open-source detection algorithms and do not fall under the category of CSI raw data.

Furthermore, we appreciate all the questions, and we will consider optimizing the explanations in the [Wi-Fi Channel State Information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information) documentation.

MacChu0315-Espressif, 2023-10-22:
Hi @Yanc14,

The issue you mentioned here can be referenced in #139.

Lijianping77, 2023-10-23:
> 嗨，
> 
> 关于CSI原始数据，每2个字节的数据对应一个子载波，第一个字节代表虚部，第二个字节代表实部。我们以“第二通道（上图）+HT + 40MHz带宽+非STBC”的场景为例。前 128 位派生自 LLTF（对应于信道中的子载波索引，范围为 -64 到 -1）。这 128 字节的数据代表 -64 子载波的虚部、-64 子载波的实部、-63 子载波的虚部，依此类推。
> 
> 至于你的第二个问题，这些数据是由非开源检测算法提供的，不属于CSI原始数据的范畴。
> 
[image]

> 此外，我们感谢所有问题，我们将考虑优化 [Wi-Fi 信道状态信息](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information)文档中的解释。
假设在这种协议下，总共也才52条子载波啊，怎么会有64条子载波呢

Lijianping77, 2023-10-23:
[image]

64条子载波，然后有12条恒为0，所以52条可用对吧？

Lijianping77, 2023-10-23:
esp32可以接收路由的两根天线的子载波吗？