espressif/esp-csi #146: The relationship between "sub-carrier index" and "sub-carrier range".

harlanho4, 2023-10-30:
This [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information) introduces the sub-carrier index under different channels and packet information. I have some questions about the relationship between sub-carrier index and sub-carrier range.

1. Assuming the secondary channel is set to none, the signal mode is non HT, the channel bandwidth is 20MHz, and STBC is non STBC, according to the table in the [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information), the sub-carrier index should be "0 ~ 31, -32 ~ 1". My understanding is that this index also represents the storage order of sub-carriers. However, in the 802.11 standard, the range of sub-carriers is -32 to 32 (including null sub-carriers), so do we need to adjust the stored sub-carrier order from "0 ~ 31, -32 ~ -1" to "-32 ~ -1, 0 ~ 31"?[image]
2. How do other sub-carrier indexes, such as "0 ~ 63"(secondary channel : below, HT, 40 MHz, non STBC), correspond to the sub-carrier range of -32 to 32? In the example [code](https://github.com/espressif/esp-csi/blob/master/examples/get-started/tools/csi_data_read_parse.py) provided by you, when removing null subcarriers, it is assumed that sub-carrier index "0 ~ 63" correspond to the sub-carrier range of -32 to 32 (null subcarrier index: -32 to -27, 0, 28 to 32). Can you provide more detailed explanations?
```python
# Remove invalid subcarriers
# secondary channel : below, HT, 40 MHz, non STBC, v, HT-LFT: 0~63, -64~-1, 384
csi_vaid_subcarrier_index = []
# LLTF: 52
csi_vaid_subcarrier_index += [i for i in range(6, 32, CSI_VAID_SUBCARRIER_INTERVAL)]
csi_vaid_subcarrier_index += [i for i in range(33, 59, CSI_VAID_SUBCARRIER_INTERVAL)]
```
3. If there is no other explanation, we may mistakenly assume that subcarrier index "-64 ~ -1" corresponds to the sub-carrier range of 32 to -32.

In addition, as stated in the [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information), if first_word_invalid field of [wifi_csi_info_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv415wifi_csi_info_t) is true, it means that the first four bytes of CSI data is invalid due to a hardware limitation in ESP32. However, in the [console_test code](https://github.com/espressif/esp-csi/blob/master/examples/esp-radar/console_test/main/app_main.c), first_word_invalid is set to 0. Why is this?
```C
len += sprintf(buffer + len, "CSI_DATA,%d,%u,%u,%s," MACSTR ",%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%d,%u,%d,%d,%d,%d,%d,",
               count++, esp_log_timestamp(), g_console_input_config.collect_number, g_console_input_config.collect_taget,
               MAC2STR(info->mac), rx_ctrl->rssi, rx_ctrl->rate, rx_ctrl->sig_mode,
               rx_ctrl->mcs, rx_ctrl->cwb, rx_ctrl->smoothing, rx_ctrl->not_sounding,
               rx_ctrl->aggregation, rx_ctrl->stbc, rx_ctrl->fec_coding, rx_ctrl->sgi,
               rx_ctrl->noise_floor, rx_ctrl->ampdu_cnt, rx_ctrl->channel, rx_ctrl->secondary_channel,
               rx_ctrl->timestamp, rx_ctrl->ant, rx_ctrl->sig_len, rx_ctrl->rx_state, info->valid_len, 0);
```

zhanzhaocheng, 2023-12-07:
@harlanho4 You can use  [wifi_csi_filtered_cb](https://github.com/espressif/esp-csi/blob/master/examples/esp-radar/console_test/main/app_main.c#L343C70-L353) obtains CSI data, and the esp radar module filters out invalid subcarriers, making it easier for you to use

1. This is determined by Wi-Fi phy
2. You can refer to this form:
    [image]
3. Due to the absence of valid CSI data in the first four bytes on certain channels, rst_word_invalid is sometimes set to 0. Among all Espressif chips supporting CSI functionality, the performance of esp32 is comparatively lower. Therefore, it is recommended to utilize the CSI functionality on ESP32-C3 or ESP32-S3.

luoruisichuan, 2023-12-31:
从官网上下载了csi_data.csv文件来查看CSI数据的格式，其中“data"字段是有效载荷信息，对应于CSI信息中各个子载波信号（复数）的实部和虚部数值，看了这个字段，共有384个元素，按照两个元素（分别为实部和虚部）对应一个子载波信号，则其对应192个子载波信号，想请教下，这384个元素（以索引来区分），是如何与192个子载波信号对应的？比如，是否data[0]和data[1]共同对应一个子载波信号？另外，192个子载波，按照WIFI规范，MIMO的每一路有64个子载波，是否这192个子载波对应3路？比如，发送天线有3根，接收天线只有1根。

在esp-csi/examples/esp-radar/console_test/tools/esp_csi_tool.py这个文件中，csi_data_handle(self, data)函数对CSI信息中子载波信息进行处理，其中line 994- line 996如下：

for i in range(CSI_DATA_COLUMNS):
        data_complex = complex(csi_raw_data[csi_vaid_subcarrier_index[i] * 2],
                               csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1])

CSI_DATA_COLUMNS=12，若i=0，csi_raw_data[csi_vaid_subcarrier_index[i]=0，则复数的实部是csi_raw_data[0]，虚部是csi_raw_data[-1]？若i=1，此时csi_raw_data[csi_vaid_subcarrier_index[i]=5，则复数的实部是csi_raw_data[10]，虚部是csi_raw_data[9]？
总之，能否介绍下csi_data.csv文件中“data"字段中各个元素与各个子载波信号信号的实部和虚部之间的对应关系？
谢谢

toniadame, 2025-04-11:
Hi @zhanzhaocheng,

Regarding question 2 from @harlanho4: let's suppose we're using *csi_vaid_subcarrier_index* from *csi_data_read_parse* to extract the indeces of the 52 data subcarriers corresponding to LLTF. Therefore, we have two buches of 26 data subcarriers: from index 6 to 31 and from index 33 to 58.

In that case, and assuming the 4 interspersed pilot subcarriers at IEEE 802.11n index (-21, -7, 7, and 21), can we consider that the following equivalence table is correct?

<small>

### First bunch of 26 data subcarriers

| IEEE 802.11n index | -28 | -27 | -26 | -25 | -24 | -23 | -22 | -21 | -20 | -19 | -18 | -17 | -16 | -15 | -14 | -13 | -12 | -11 | -10 | -9  | -8  | -7  | -6  | -5  | -4  | -3  | -2  | -1  |
|-------------------------------|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|
| Espressif index |  6  |  7  |  8  |  9  | 10  | 11  | 12  |     | 13  | 14  | 15  | 16  | 17  | 18  | 19  | 20  | 21  | 22  | 23  | 24  | 25  |     | 26  | 27  | 28  | 29  | 30  | 31  |

### Second bunch of 26 data subcarriers

| IEEE 802.11n index | 1   | 2   | 3   | 4   | 5   | 6   | 7   | 8   | 9   | 10  | 11  | 12  | 13  | 14  | 15  | 16  | 17  | 18  | 19  | 20  | 21  | 22  | 23  | 24  | 25  | 26  | 27  | 28  |
|-------------------------------|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|-----|
| Espressif index  | 33  | 34  | 35  | 36  | 37  | 38  |     | 39  | 40  | 41  | 42  | 43  | 44  | 45  | 46  | 47  | 48  | 49  | 50  | 51  |     | 52  | 53  | 54  | 55  | 56  | 57  | 58  |

</small>

[image]

Thanks in advance.