espressif/esp-csi #150: Not an issue but a question

abhi-3001, 2023-12-05:
In the setup 2 [CSI Between Devices], does the device1 ping device2 or router changes the their response with each other??

zhanzhaocheng, 2023-12-07:
@abhi-3001 I didn't quite understand your question. Can you provide a more detailed description?

abhi-3001, 2023-12-14:
Hello @zhanzhaocheng , actually I want to build the setup 2, where the CSI data between devices are share. But I didn’t understand that how the devices are sharing the CSI data and how it is independent from the router. 

Does the Device1 ping the Device2 and getting the CSI data??  If not then, please explain your method to sharing the CSI data between devices.

I will wait for your response. Thank you!

MacChu0315, 2023-12-14:
Hi @abhi-3001, whenever either device pings the router, both devices can listen to the ICMP reply packets from the router. In order to ensure that the acquired CSI data originates from the target router, MAC address filtering can be applied to the received packets.

MacChu0315-Espressif, 2023-12-14:
Hi @abhi-3001, whenever either device pings the router, both devices can listen to the ICMP reply packets from the router. In order to ensure that the acquired CSI data originates from the target router, MAC address filtering can be applied to the received packets.

4444monkey, 2024-01-03:
Regarding the aforementioned question, setup 2 can be realized by deploying 2 ESP32 devices. Does it mean just using 2 devices and ping the same router simultaneously? Both devices are burned via sub-project csi_recv_router. Is that correct?
Wanna know the mechanism and how to apply it?

MacChu0315-Espressif, 2024-01-04:
Your description is essentially correct. You can use two ESP devices to ping the same router simultaneously.

4444monkey, 2024-01-04:
The CSI data flow will come from router passing to each ESP device, and I just collect the data either from  ESP device A or B, right? As stated in README, setup2 will stabilize the CSI data collection, correct? In other words, It means wherever the router was equipped, data collected by using setup2 will be more trustworthy than just using one ESP device, correct?

MacChu0315-Espressif, 2024-01-07:
It seems there is a bit of misunderstanding in our interpretation.

Firstly, regarding your previous description of "the CSI data between devices are share": Due to the fact that CSI data is a result calculated by the receiving end, similar to RSSI, device A cannot directly obtain the CSI data between device B and its transmitter (in this case, the router), unless the other device transfers already parsed CSI data to it.

However, even though device A cannot acquire the CSI data between device B and its transmitter, they can share the same transmitter. In the current `csi_recv_router` sub-project, there are two main functionalities: 1. Sending pings to the target router, and 2. Capturing all airborne packets (without filtering) and analyzing the CSI data between the ESP and the transmitter. Therefore, it is possible for device A to ping the router while device B simultaneously listens for responses from the same router (ICMP). In this scenario, whether the ping to the router functionality is enabled on device B mainly affects the frequency of device B obtaining CSI data. Since device A is consistently pinging the router, device B can also acquire CSI data.

Therefore, it might not be accurate to say that the obtained CSI data are more trustworthy in this situation. I hope this clarifies the matter. If you have any further questions, please feel free to reach out to us.

4444monkey, 2024-01-07:
Your explanation is clear.  I just wanna dig into more about the advantages of setup2. Why say the "Does not depend on the location of the router"? Based on my previous understanding, I thought the location of the router would have less impact using setup2. How can I understand it?

MacChu0315-Espressif, 2024-01-15:
The description in the Readme was written quite some time ago, and there might be some misleading information here. We believe the advantage of this approach lies in the fact that, for Device A, besides receiving CSI data by responses from the router, it can simultaneously obtain CSI data by pings from Device B and responses from the router to Device B. Additionally, if A does not send external ping packets, it can reduce power consumption while still receiving CSI data between itself and the router.

4444monkey, 2024-01-15:
Now, I completely figure out how it works now. Thx~