espressif/esp-csi #151: The CSI test has not result

Joker-kill, 2023-12-06:
According to the script 'python esp_csi_tool.py -p COM7', when we run the script, there is not any result as following,
[image]

but the correct result is as following,

[image]

what is problem?
thanks very much for your help.

zhanzhaocheng, 2023-12-07:
@Joker-kill Is the device running properly? Does the device's lighting change when you move?, Could you provide the device logs when you're not running this script?

sebasrpalacio95, 2023-12-12:
Hi i have the same issue, the device lighting change when i move but i have not data :(

MacChu0315-Espressif, 2023-12-12:
@sebasrpalacio95 May I ask what devices you are using, and at what baud rate the serial port is configured?

huaxiuyiwei, 2024-03-23:
注意一下波特率，默认波特率是2000000，可以修改成115200，注意波特率要统一，不然就没有log
[image]

rav2n-dev, 2024-04-11:
Same trouble here as well