espressif/esp-csi #153: The query about raw CSI data manifestation

4444monkey, 2023-12-12:
Hi There,
When I tried the sub-tasks "csi_recv_router" on different routers.
They just responded CSI raw data in the different format layouts as follows (I skipped other information):
1. CSI_DATA, e8:48:b8:e9:2b:b1:[0,0,0,0,0,0,0,0,0,0,0,0,1,-3,2,-2,3,-2,4,-1,4,-1,5,0,5,0,6,1,6,1,6,1,5,2,5,2,4,3,3,3,3,4,2,5,1,6,0,6,-1,7,-1,7,-2,8,-2,8,-2,8,-2,7,-3,6,-3,5,0,0,-4,3,-4,1,-5,0,-6,-2,-7,-2,-8,-3,-9,-3,-10,-3,-11,-3,-12,-3,-12,-2,-11,-1,-10,-1,-9,0,-7,0,-5,0,-3,-1,-1,-2,1,-3,3,-5,4,-8,4,-10,5,-12,4,-14,3,-16,2,-17,0,0,0,0,0,0,0,0,0,0]

2.CSI_DATA,54:b8:0a:7e:62:64:
[0,0,13,-6,11,-6,8,-5,6,-5,2,-5,-1,-6,-4,-7,-6,-8,-8,-9,-10,-10,-12,-12,-12,-13,-13,-14,-13,-14,-12,-15,-11,-15,-10,-14,-8,-13,-7,-11,-5,-8,-5,-5,-5,-2,-5,1,-6,4,-8,6,-9,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2,6,5,8,6,11,7,13,7,15,7,17,6,18,5,18,4,18,2,17,1,15,1,13,1,11,2,8,2,4,4,2,5,-1,7,-3,9,-5,11,-7,12,-8,14,-9,15,-9,17,-9,17,-9,16,-8]"

I was wondering why I got these two different data patterns since the only difference was the router.

Parts of the null subcarriers are at the start, middle, and end based on my understanding. However, I was not able to figure out the second one. Seems the data is still fine but different sequences, and the null subcarriers line in the middle of the data packet. What is the possible reason for that? Cuz I won't know where I should start to compute for the second one.

Hope someone can give me a guide for it.

MacChu0315-Espressif, 2023-12-14:
The format of CSI data from wireless packets with different channels and bandwidths may vary. To obtain well-organized CSI data, the 'valid_data' field in the 'wifi_csi_filtered_info_t' structure can be utilized.

4444monkey, 2023-12-14:
Do you mean the function in "/esp-csi/tree/master/examples/esp-radar/console_test/main)
/app_main.c"?

MacChu0315-Espressif, 2023-12-14:
yes, exactly.

4444monkey, 2023-12-15:
For the order of the CSI raw values I acquired, were they collected in sequence like subcarrier 1,  subcarrier 2, ... or not?
I am not quite sure about the order of the data array.
Or I have to put another rule to sort them?

MacChu0315-Espressif, 2023-12-19:
Hi @4444monkey ,
The issue you mentioned here can be referenced in https://github.com/espressif/esp-csi/issues/139.

4444monkey, 2023-12-19:
So in light of the question I lised above, there were only 52 valid subcarriers over the acquired 64 subcarriers regardless of the order and the data packet varies from the PHY standard.
If I have to process the CSI data analysis further, I may decide the order among them, correct?
Just hope I catch the right point about that and thanks for your reply.

MacChu0315-Espressif, 2023-12-21:
> So in light of the question I listed above, there were only 52 valid subcarriers (104 bytes) over the acquired 64 subcarriers (128 byte) regardless of the order and the data packet varies from the PHY standard. If I have to process the CSI data analysis further, I may decide the order among them like subcarrier 1, 2, 3, ..., correct? Just hope I catch the right point about that since not much info about CSI order in that discussion. Thanks for your reply.

Got it, we will reach the internal R&D to confirm the order of subcarrier among CSI data with different formats and keep you informed of any progress.

4444monkey, 2023-12-21:
Splendid! Thanks for your support.

TS-Sun-P, 2024-04-19:
May I answer your latest progress on this issue?I also want to know the order of subcarrier among CSI data with different formats.Thanks for your reply.

TS-Sun-P, 2024-04-19:
May you answer the latest progress on this issue?I also want to konw the order of subcarrier among CSI data with different formats.Thanks for your reply.

xinxue123, 2024-06-17:
May you answer the latest progress on this issue?I also want to konw the order of subcarrier among CSI data with different formats.Thanks for your reply.