espressif/esp-csi #154: The issue of CSI monitoring CSV file creation

4444monkey, 2023-12-26:
Hi there,

Somehow I try to use python generating the CSV file, but it showed the following message. How can I resolve this issue and successfully get the CSV file? I'm using VScode ESP-IDF version 5.1.2 Windows-based on my project.
Traceback (most recent call last):
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\ansi_color_converter.py", line 72, in _output_write
    self.output.write(data.decode())  # type: ignore
                      ^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xef in position 0: invalid continuation byte

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\ENG\esp\esp-idf\tools\idf.py", line 779, in <module>
    main()
  File "C:\Users\ENG\esp\esp-idf\tools\idf.py", line 714, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\click\core.py", line 1128, in __call__   
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\click\core.py", line 1053, in main       
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\click\core.py", line 1691, in invoke     
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\click\core.py", line 1628, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\click\core.py", line 754, in invoke      
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ENG\esp\esp-idf\tools\idf.py", line 611, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "C:\Users\ENG\esp\esp-idf\tools\idf.py", line 193, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "C:\Users\ENG\esp\esp-idf\tools\idf_py_actions\serial_ext.py", line 167, in monitor
    RunTool('idf_monitor', monitor_args, args.project_dir, build_dir=args.build_dir, hints=hints, interactive=True, convert_output=True)()
  File "C:\Users\ENG\esp\esp-idf\tools\idf_py_actions\tools.py", line 304, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "asyncio\runners.py", line 190, in run
  File "asyncio\runners.py", line 118, in run
  File "asyncio\base_events.py", line 653, in run_until_complete
  File "C:\Users\ENG\esp\esp-idf\tools\idf_py_actions\tools.py", line 348, in run_command
    await asyncio.gather(
  File "C:\Users\ENG\esp\esp-idf\tools\idf_py_actions\tools.py", line 421, in read_and_write_stream
    output_converter.write(output)
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\ansi_color_converter.py", line 131, in write
    self._output_write(b)
  File "C:\Users\ENG\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\ansi_color_converter.py", line 89, in _output_write
    self.output.write(self.decode_buffer.decode(errors='replace'))  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'cp950' codec can't encode character '\ufffd' in position 0: illegal multibyte sequence
Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x0000021C664A7600>
Traceback (most recent call last):
  File "asyncio\base_subprocess.py", line 125, in __del__
  File "asyncio\base_subprocess.py", line 78, in __repr__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x0000021C664CCD60>
Traceback (most recent call last):
  File "asyncio\proactor_events.py", line 116, in __del__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe

MacChu0315-Espressif, 2023-12-26:
It seem that this error doesn't come from neither CSI repo nor IDF repo, but more related to the string you save. Try to figure out whether invalid UTF-8 sequences is included may help.

4444monkey, 2023-12-26:
I might still not get it. I used the sub project per "examples/get-started/csi_recv" and try to save data in the format of CSV file using "idf.py monitor | findstr "CSI_DATA" > esp-csi.csv ". Then, came out the mentioned error code above. Do you mean the python tool cannot decode the exported CSI data format by default? Or any action I have to execute?

MacChu0315-Espressif, 2023-12-26:
Strings come from the terminal may contain invalid formats that are not universally supported.  I conducted an experiment to illustrate this feature.

**Step 1:** Save the terminal input to a file named `CSI.txt` using the command "script -f ~/Desktop/CSI.txt".
[image]

**Step 2:** Format `CSI.txt` into a CSV file using the command "cat ~/Desktop/CSI.txt | grep 'CSI_DATA' > ~/Desktop/CSI.csv". The resulting content is as follows:
[image]

**Step 3:** Copy the strings from `CSI.txt` and paste them into `CSI_1.txt` using "Ctrl+C" and "Ctrl+V". The resulting content is shown below:
[image]

**Step 4:** Repeat the commands from Step 2. The outcome is as follows:
[image]

For pasting strings into text files, the formatting of the strings is automatically removed. Hope that i've pointed the cause of the issue out.

4444monkey, 2023-12-26:
Thanks for your illustration in detail. However, I intend to acquire the CSI data saved in the format of CSV (real-time) while the serial port keeps outputting the CSI data packet. Now, I'm a little bit overwhelmed. I have no idea how to deal with that since no information can be referred to as well. If you have tips for that, it will be better. Thanks again for your clear explanation.

MacChu0315-Espressif, 2023-12-27:
The reason why the commands above can't working functionally is that this command is supportted by ubuntu terminal.
If you want to save CSI data in CSV real-time, you can simplily use esp_radar/console_test/tools/esp_csi_tool.py or you can write your own python script. Your script opens the serial to get the outputted data from esp device and save the strings with 'csv.writer' command. You can refer to the codes in serial_handle from esp_csi_tool.py.

4444monkey, 2023-12-27:
It works thx! I did use the Python tools as you suggested. I successfully got the output CSV data. Maybe, the result of the error I encountered was that the required Python tool lists stated in the text file were not installed, or my anti-virus software blocked the Python tool version upgrade. Thanks for your tips.

closed 2023-12-27 by 4444monkey