espressif/esp-csi #156: The relation between the element of CSI data payload and sub-carrier index

luoruisichuan, 2024-01-01:
从官网上下载了csi_data.csv文件来查看CSI数据的格式，其中“data"字段是有效载荷信息，对应于CSI信息中各个子载波信号（复数）的实部和虚部数值，看了这个字段，共有384个元素，按照两个元素（分别为实部和虚部）对应一个子载波信号，则其对应192个子载波信号，想请教下，这384个元素（以索引来区分），是如何与192个子载波信号对应的？比如，是否data[0]和data[1]共同对应一个子载波信号？另外，192个子载波，按照WIFI规范，MIMO的每一路一般有64个子载波，是否这192个子载波对应3路？比如，发送天线有3根，接收天线只有1根。

在esp-csi/examples/esp-radar/console_test/tools/esp_csi_tool.py这个文件中，csi_data_handle(self, data)函数对CSI信息中子载波信息进行处理，其中line 994- line 996如下：

for i in range(CSI_DATA_COLUMNS):
        data_complex = complex(csi_raw_data[csi_vaid_subcarrier_index[i] * 2],
                               csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1])

CSI_DATA_COLUMNS=12，若i=0，csi_raw_data[csi_vaid_subcarrier_index[i]=0，则复数的实部是csi_raw_data[0]，虚部是csi_raw_data[-1]？若i=1，此时csi_raw_data[csi_vaid_subcarrier_index[i]=5，则复数的实部是csi_raw_data[10]，虚部是csi_raw_data[9]？
总之，能否介绍下csi_data.csv文件中“data"字段中各个元素与各个子载波信号信号的实部和虚部之间的对应关系？
谢谢

MacChu0315-Espressif, 2024-01-03:
Part of your questions can refer to former issues as listed: #103 / #114 / #145.
And ESP32 doesn't support MIMO by now, it can receive only one wireless packet at a time. Discriptions on subcarriers of 384 elememts can refer to [Wi-Fi Channel State Information]{https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information}.

Gs-Smit, 2024-01-03:
> 您的部分问题可以参考列出的先前问题：#103 / #[114 / #](https://github.com/espressif/esp-csi/issues/114)[145](https://github.com/espressif/esp-csi/issues/145)。而 ESP32 目前还不支持 MIMO，一次只能接收一个无线数据包。384 elememts 的子载波的描述可以参考 [Wi-Fi 信道状态信息]{[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information}](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information%7D)。

参考你所提及的：2个元素（虚部，实部）对应一个子载波，那就data_complex = complex（csi_raw_data[csi_vaid_subcarrier_index[i] * 2]，csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1]）应该改为
data_complex = complex（csi_raw_data[csi_vaid_subcarrier_index[i] * 2 +1 ]，csi_raw_data[csi_vaid_subcarrier_index[i] * 2 ]）才对。是代码出错了，还是我理解错了？

luoruisichuan, 2024-01-03:
I still have some problem.
以 issue #145中数据为例,  

CSI_DATA,774,a2:6b:f8:4a:4a:87,-20,11,1,7,0,1,1,0,1,0,1,-97,0,1,0,13230366,0,83,0,128,0,"
[0,0,2,5,5,5,7,8,9,11,10,10,11,11,12,11,11,10,12,12,11,12,10,10,8,6,6,4,5,2,3,0,1,-3,0,-6,-1,-9,-1,-12,-2,-15,-2,-17,-2,-15,-2,-16,-1,-16,1,-18,2,-16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,15,-8,16,-8,13,-7,13,-6,12,-5,14,-3,11,-2,9,-2,5,-1,1,-2,-2,-2,-5,-3,-8,-4,-10,-4,-12,-5,-15,-7,-16,-8,-17,-8,-15,-6,-15,-6,-14,-5,-13,-5,-12,-5,-9,-2,-6,-1,-4,0]"

CSI raw data starts from "[" and ends with "]", hence the CSI raw data is as follows,

[0,0,2,5,5,5,7,8,9,11,10,10,11,11,12,11,11,10,12,12,11,12,10,10,8,6,6,4,5,2,3,0,1,-3,0,-6,-1,-9,-1,-12,-2,-15,-2,-17,-2,-15,-2,-16,-1,-16,1,-18,2,-16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,15,-8,16,-8,13,-7,13,-6,12,-5,14,-3,11,-2,9,-2,5,-1,1,-2,-2,-2,-5,-3,-8,-4,-10,-4,-12,-5,-15,-7,-16,-8,-17,-8,-15,-6,-15,-6,-14,-5,-13,-5,-12,-5,-9,-2,-6,-1,-4,0]

列表里共有128个元素，一个子载波对应2个元素，那么这个列表里的元素对应64条子载波。列表里各个元素与子载波信号的具体对应关系，以列表中前8个元素“0,0,2,5,5,5,7,8”为例，是不是如下关系：
0,0对应一条子载波；
2,5对应一条子载波；
5,5对应一条子载波；
7,8对应一条子载波。
并且，在最后一条子载波中，“7”是信号的虚部，“8”是信号的实部？
因为需要根据CSI raw data分别计算各个子载波信号的幅度和相位，麻烦看看以上所述例子中，各个元素与子载波信号的对应关系是否正确，如果不正确，以上述例子数据为例，正确的对应关系应该是怎么样。
thanks for your help very much!

luoruisichuan, 2024-01-04:
如果按照github上代码，

for i in range(CSI_DATA_COLUMNS):
data_complex = complex(csi_raw_data[csi_vaid_subcarrier_index[i] * 2],
csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1])

以前述列表中前8个元素“0,0,2,5,5,5,7,8”为例

应该是第一个0和整个列表最后一个元素配对，即0，0，前面一个0是列表最后一个元素
第2个0与2配对，即0，2，对应一条子载波
5和5配对，对应一条子载波
5和7配对，对应一条子载波
第8个元素8，应该和第9个元素9配对，对应一条子载波

到底是哪种配对方式呢？
thanks for your help very much!

nighthink, 2024-01-10:
> 如果按照github上代码，
>
> for i in range(CSI_DATA_COLUMNS): data_complex = complex(csi_raw_data[csi_vaid_subcarrier_index[i] * 2], csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1]) [...]

I have this question as well.

MacChu0315-Espressif, 2024-01-15:
> > 您的部分问题可以参考列出的先前问题：#103 / #[114 / #](https://github.com/espressif/esp-csi/issues/114)[145](https://github.com/espressif/esp-csi/issues/145)。而 ESP32 目前还不支持 MIMO，一次只能接收一个无线数据包。384 elememts 的子载波的描述可以参考 [Wi-Fi 信道状态信息]{[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information}](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information%7D)。
> 
> 参考你所提及的：2个元素（虚部，实部）对应一个子载波，那就data_complex = complex（csi_raw_data[csi_vaid_subcarrier_index[i] * 2]，csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1]）应该改为 data_complex = complex（csi_raw_data[csi_vaid_subcarrier_index[i] * 2 +1 ]，csi_raw_data[csi_vaid_subcarrier_index[i] * 2 ]）才对（i从0开始）。是代码出错了，还是我理解错了？

Hi, @luoruisichuan, @Gs-Smit and @nighthink, your understanding was correct. There are indeed some issues in the code responsible for displaying the UI. We plan to address and fix these issues in the near future. Thank you for bringing this to our attention.

nighthink, 2024-01-15:
> Hi, @luoruisichuan, @Gs-Smit and @nighthink, your understanding was correct. There are indeed some issues in the code responsible for displaying the UI. We plan to address and fix these issues in [...]

Thank you so much for your reply, it helped me a lot!

closed 2024-01-16 by EspHuifeng