espressif/esp-csi #157: csi run through mdf

Connor-WangK, 2024-01-03:
can CSI detection run through mdf topology,
I want to design a system for underground garage lighting, 
each light communicate with other nodes through mdf, 
if some one deteact a car or body is moving, it will notify other node to turn up the brightness.

MacChu0315-Espressif, 2024-01-03:
Can you clarify the meaning of MDF, for example, whether it refers to [esp-mdf](https://github.com/espressif/esp-mdf)?

Connor-WangK, 2024-01-04:
yes, I means esp-mdf, maybe esp-now is ok.

MacChu0315-Espressif, 2024-01-08:
You can obtain CSI data and esp-radar results not only from esp-mdf but also from other esp-now related repositories after integrating the ESP-CSI repository into these projects. However, it is not recommended to develop new projects using ESP-MDF, as it is no longer actively maintained. Instead, we suggest using [ESP-MESH-LITE](https://github.com/espressif/esp-mesh-lite), which is a more recommended alternative.

Connor-WangK, 2024-01-10:
> You can obtain CSI data and esp-radar results not only from esp-mdf but also from other esp-now related repositories after integrating the ESP-CSI repository into these projects. However, it is not [...]

I have tried both 'esp-mdf' and 'esp-mesh-lite' in a case which link for roadlights, the 'esp-mdf' is better,
I guess 'esp-mesh-lite' is suitable work as a less nodes and short distance network topology.

MacChu0315-Espressif, 2024-01-15:
ESP-MDF is built on a proprietary WiFi-mesh protocol, making issue identification more challenging. In contrast, ESP-MESH-LITE is a repository founded on softAP and station, actively maintained, and is the recommended choice for development due to its ease of use and support.