espressif/esp-csi #158: wifi_csi_cb cause panic in console_test

Oreobird, 2024-01-04:
Use two ESP32-C3 to run the steps within https://github.com/espressif/esp-csi/blob/master/examples/esp-radar/console_test/README.md and encounter error as below.  Software version is esp-idf_v5.0 and the latest clone from esp-csi master.
`I (35) boot: ESP-IDF 264e1242-dirty 2nd stage bootloader
I (35) boot: compile time 14:47:07
I (35) boot: chip revision: v0.3
I (35) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32c3: SPI Speed      : 80MHz
I (36) boot.esp32c3: SPI Mode       : QIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (36) boot: Enabling RNG early entropy source...
I (37) boot: Partition Table:
I (37) boot: ## Label            Usage          Type ST Offset   Length
I (37) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (38) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (38) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (38) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (39) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (39) boot:  5 ota_0            OTA app          00 10 00020000 001ca000
I (40) boot: End of partition table
I (40) esp_image: segment 0: paddr=00020020 vaddr=3c0a0020 size=22978h (141688) map
I (60) esp_image: segment 1: paddr=000429a0 vaddr=3fc91a00 size=02bach ( 11180) load
I (63) esp_image: segment 2: paddr=00045554 vaddr=40380000 size=0aac4h ( 43716) load
I (70) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=9aa58h (633432) map
I (159) esp_image: segment 4: paddr=000eaa80 vaddr=4038aac4 size=06dach ( 28076) load
I (164) esp_image: segment 5: paddr=000f1834 vaddr=50000010 size=00010h (    16) load
I (170) boot: Loaded app from partition at offset 0x20000
I (170) boot: Disabling RNG early entropy source...
I (171) cpu_start: Pro cpu up.
I (179) cpu_start: Pro cpu start user code
I (180) cpu_start: cpu freq: 160000000 Hz
I (180) cpu_start: Application information:
I (180) cpu_start: Project name:     console_test
I (180) cpu_start: App version:      e91f136-dirty
I (180) cpu_start: Compile time:     Jan  4 2024 14:47:02
I (181) cpu_start: ELF file SHA256:  4d31ad02283d16cb...
I (181) cpu_start: ESP-IDF:          264e1242-dirty
I (181) heap_init: Initializing. RAM available for dynamic allocation:
I (182) heap_init: At 3FC98B80 len 00043B90 (270 KiB): DRAM
I (182) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (182) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (183) spi_flash: detected chip: generic
I (184) spi_flash: flash io: qio
W (184) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (184) cpu_start: Starting scheduler.

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
csi> I (506) pp: pp rom version: 9387209
I (507) net80211: net80211 rom version: 9387209
I (508) wifi:wifi driver task: 3fca3c24, prio:23, stack:6656, core=0
I (508) system_api: Base MAC address is not set
I (509) system_api: read default base MAC address from EFUSE
I (511) wifi:wifi firmware version: 0d470ef
I (512) wifi:wifi certification version: v7.0
I (512) wifi:config NVS flash: enabled
I (512) wifi:config nano formating: disabled
I (513) wifi:Init data frame dynamic rx buffer num: 32
I (514) wifi:Init management frame dynamic rx buffer num: 32
I (514) wifi:Init management short buffer num: 32
I (515) wifi:Init dynamic tx buffer num: 32
I (515) wifi:Init static tx FG buffer num: 2
I (516) wifi:Init static rx buffer size: 2212
I (516) wifi:Init static rx buffer num: 10
I (517) wifi:Init dynamic rx buffer num: 32
I (518) wifi_init: tcpip mbox: 32
I (519) wifi_init: udp mbox: 6
I (519) wifi_init: tcp mbox: 6
I (520) wifi_init: tcp tx win: 5744
I (520) wifi_init: tcp rx win: 5744
I (520) wifi_init: tcp mss: 1440
I (521) wifi_init: WiFi IRAM OP enabled
I (521) wifi_init: WiFi RX IRAM OP enabled
I (522) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (553) wifi:mode : null
I (554) wifi:Set ps type: 0

I (555) wifi:ic_enable_sniffer
I (556) esp_radar: CSI_RADAR Version: 0.1.0, Compile time: Jun  6 2023 - 14:24:26
I (557) esp_radar: csi_recv_interval: 10, csi_handle_time: 250, csi_handle_window: 50, csi_handle_buffer: 60
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x400587ec  RA      : 0x4200ba48  SP      : 0x3fca3a70  GP      : 0x3fc92200  
0x4200ba48: wifi_csi_cb at /Users/zhanzhaocheng/project/esp-csi/esp-csi-ext/examples/console_test/components/esp-csi/src/esp_radar.c:729 (discriminator 3)

TP      : 0x3fc81b84  T0      : 0x87002c80  T1      : 0x3b19c130  T2      : 0x801e4bd5  
S0/FP   : 0x3fca3d7c  S1      : 0x00000040  A0      : 0x3fc9adad  A1      : 0x0000001d  
A2      : 0x00000034  A3      : 0x00000001  A4      : 0x3fc9ade1  A5      : 0x3fc9adad  
A6      : 0x00000000  A7      : 0x0000001a  S2      : 0x3fc988c8  S3      : 0x3c0a79f4  
S4      : 0x3fcab6c8  S5      : 0x0000000c  S6      : 0x00000034  S7      : 0x3c0a79d8  
S8      : 0x3c0a7000  S9      : 0x3fce0000  S10     : 0x00000001  S11     : 0x3fce0000  
T3      : 0x1810d8a0  T4      : 0x941fe3f7  T5      : 0x00085981  T6      : 0x982b0000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x0000001d  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x400587ec in ?? ()
#0  0x400587ec in ?? ()
#1  0x4200ba48 in wifi_csi_cb (ctx=<optimized out>, info=0x3fca3d7c) at /Users/zhanzhaocheng/project/esp-csi/esp-csi-ext/examples/console_test/components/esp-csi/src/esp_radar.c:728
#2  0x2e617473 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 4d31ad02283d16cb
`

MacChu0315-Espressif, 2024-01-15:
We couldn't reproduce it on ESP-IDF v5.0 and esp-csi master. Also, we noticed in your logs that the IDF version is mentioned as ESP-IDF 264e1242-dirty. Checking your IDF version is recommended.

Oreobird, 2024-01-15:
Thx, using v5.1.1 works for me now

closed 2024-01-15 by Oreobird