espressif/esp-csi #161: uxPriority assert fails in tasks.c for console_test example

James-Rhodes, 2024-01-26:
Hi, 

When trying to run the example in examples/esp-radar/console_test the example fails with the message 

`assert failed: prvInitialiseNewTask tasks.c:1061 (uxPriority < ( 25 ))`

I investigated esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c and found that commenting out the assert allowed the example to run as expected. I assume the issue is that whatever priority you are passing in for one of the tasks is very large (greater than 25) and this is causing issues. I am not sure if this is platform-specific as my specific configMAX_PRIORITIES was set to 25 but nonetheless, I think that this can be fixed within your source code. 

Thanks!

MacChu0315-Espressif, 2024-01-29:
To assist you more effectively in resolving your issues, kindly specify the versions of both esp-csi and esp-idf that you are currently using. If feasible, please share the commit number, as this will aid us in pinpointing and addressing the problem with greater precision.

MacChu0315-Espressif, 2024-01-30:
We have also conducted some tests, and it seems that you are using idf-master. We will be adapting to it in the near future. In the meantime, we recommend pairing it with **IDF-v5.1/5.0/4.4** for optimal usage with this repository.

James-Rhodes, 2024-01-30:
Hi, thanks so much for getting back to me. I have confirmed I am using V5.3 of the IDF so that is almost certainly the issue. I will close this off and await the updates to this repo :)

closed 2024-01-30 by James-Rhodes