espressif/esp-csi #163: CSI DATA CONSOLE TEST PRINT & MANIPULATION

sebasrpalacio95, 2024-01-30:
Good day, I've been trying to set up the console_test example to extract only the CSI data and display it in the serial terminal since it's filtered unlike the other examples. I've really tried a lot, but haven't been able to succeed. Could you help me? Thank you very much.

MacChu0315-Espressif, 2024-01-30:
Are you experiencing compilation failures or continuous crashes and restarts? If so, please try using this demo with IDF v5.1/v5.0/v4.4 for better compatibility.

sebasrpalacio95, 2024-01-30:
No, compilation it's ok. What I mean is that I want to exclude everything related to the Python graphical environment in that example. All I want from that example (console_test) is to print the CSI data. I want to modify it to use the connect example to connect to my router and print CSI data, similar to the CSI_RECV_ROUTER example. However, in that example, I have 128 unfiltered data points, unlike the filtered data in console_test. Despite my attempts to modify it, I've been unable to get the data to appear in the serial terminal.

MacChu0315-Espressif, 2024-01-30:
No need to modify the code you cloned. All you need to do is to feed commands to the chip through the terminal. Try to feed commands like:
1. radar --csi_output_type LLFT --csi_output_format base64
2. wifi_config --ssid "TP-LINK_Liu" --password 11112222
BTW, don’t forget to modify the ssid and password in the command!

sebasrpalacio95, 2024-01-30:
Hello, Thank you very much for your response! What you're saying is very interesting; it does work indeed. However, you see... with this same example and the Python user interface, I collected a dataset to which I applied a machine learning process with TensorFlow. I now have my model in .cc. The idea is to take the CSI data from this Console_Test example to input them into the tensor and obtain predictions. That's why I need to manipulate them.