espressif/esp-csi #164: How to calculate the predict_someone_threshold and predict_move_threshold

zhaofz99, 2024-02-20:
Hello, thank you very much for sharing. Could you please explain how predict_someone_threshold and predict_move_threshold are calculated? I can't find related code from this repo. If that's not open-source, could you please give some hints about its implementation? Thank you very much!

zhanzhaocheng, 2024-02-21:
This is determined by calibration when no one is in the room, cf:[Calibration threshold](https://github.com/espressif/esp-csi/tree/master/examples/esp-radar/console_test#52-calibration-threshold)

zhaofz99, 2024-02-21:
Yes, I mean what’s the calibration algorithm behind it?

Thank you very much!

zhanzhaocheng, 2024-02-21:
**Human Motion Detection**: Calculates the maximum CSI waveform when no one is moving.
**Human Presence Detection**: Records the CSI waveform when no one is present.

zhaofz99, 2024-02-22:
Could you please give me some hints about how to transform the CSI amplitude waveform to the wander/jitter waveform on the right in ESP_CSI_TOOL? What's the unit of the wander(someone) threshold and jitter(move) threshold?

zhanzhaocheng, 2024-02-22:
@zhaofz99 I have written a related patent before, you can refer to it: [Method and Smart Device for Wi-Fi Human Detection](https://analytics.zhihuiya.com/patent-view/abst?limit=100&q=espressif%20csi&efq=ANCS%3A%28%22ESPRESSIF%20SYST%20SHANGHAI%22%29&_type=query&patentId=2528b8ad-a077-435b-a7b9-d0d302ac6c04&sort=isddesc&rows=100&page=1&source_type=search_result)