espressif/esp-csi #165: How to use router to send CSI data

2heYu110, 2024-02-29:
Good day. I am using ESP32 as my receiver and I would like to use D-Link AX3000 router to send CSI data.
I have done **3.2 Receive Wi-Fi CSI** to my ESP32 and I am able to open the CSI visualization interface. Can I know how to setup the router to send CSI data? Meanwhile, what does the phase "The router is not connected to other intelligent devices" means?
Thank you.

zhanzhaocheng, 2024-02-29:
You only need to refer to [this](https://github.com/espressif/esp-csi/blob/master/examples/esp-radar/console_test/README.md#4-interface-introduction) to the router to receive CSI data, without modifying any configuration of the router

2heYu110, 2024-02-29:
Is this the correct way to use the ESP CSI Tool? After I entered the ssid and password in the router connection window, there aren't anything shown in the "log print window" and the "CSI data waveform display window".

[image]

[image]

1dot6, 2024-03-14:
> Is this the correct way to use the ESP CSI Tool? After I entered the ssid and password in the router connection window, there aren't anything shown in the "log print window" and the "CSI data [...]

Did you figure it out? I am facing the same issue.

rav2n-dev, 2024-04-11:
The same here. No response on the gui or the console