espressif/esp-csi #168: ssert failed: prvInitialiseNewTask tasks.c:1061 (uxPriority < ( 25))

JosephJwamier, 2024-03-20:
when flash the project its give me the uxPriority in line when the radar start in the main app (esp_radar_start();)

 ssert failed: prvInitialiseNewTask tasks.c:1061 (uxPriority < ( 25))

Backtrace: 0x400819c2:0x3ffceb50 0x40089de1:0x3ffceb70 0x400916ed:0x3ffceb90 0x4008bdae:0x3ffcecb0 0x4008daaa:0x3ffcece0 0x400da334:0x3ffced20
0x400819c2: panic_abort at C:/Users/PC/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
0x40089de1: esp_system_abort at C:/Users/PC/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x400916ed: __assert_func at C:/Users/PC/esp/v5.2.1/esp-idf/components/newlib/assert.c:81
0x4008bdae: prvInitialiseNewTask at C:/Users/PC/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:1061 (discriminator 1)
0x4008daaa: xTaskCreatePinnedToCore at C:/Users/PC/esp/v5.2.1/esp-idf/components/freertos/esp_additions/freertos_tasks_c_additions.h:284
0x400da334: xTaskCreate at /Users/zhanzhaocheng/project/esp-csi/esp-csi_release/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:451 (discriminator 21)
 (inlined by) esp_radar_start at /Users/zhanzhaocheng/project/esp-csi/esp-csi-ext/examples/console_test/components/esp-csi/src/esp_radar.c:834 (discriminator 21)

mitchkapa, 2024-03-21:
I am having the same issue. I am attempting to use an ESP32 not ESP32-S3.  The CSI-SEND device is also an ESP32.  I was not clear if an ESP32-S3 is required for this example to function.

MacChu0315-Espressif, 2024-04-01:
Issue #161 may help.

hugovantriet, 2024-06-24:
I had the same issue, downgrading to the ESP-IDF version 5.0.4 fixed the problem for me.

Suvashan1, 2025-02-18:
Greetings

I am using v5.2.1 of the ESP-IDF, and trying to run FreeRTOS. I am unable to create a simple task that blinks an LED. I get this error. Is there any fix for this besides downgrading the ESP-IDF version?

NingYuxiang, 2025-02-24:
您可以暂时参照 https://github.com/espressif/esp-csi/pull/223/files 进行修改，并执行
```c
idf.py fullclean
rm -rf build
```
之后重新编译运行