espressif/esp-csi #170: 关于相位信息

Worange12, 2024-04-02:
所得到的csv文件中data中的数据格式是单数列为实数，双数列为虚数吗？如果是的也就意味着板载WiFi天线也能得到相位信息是吗？
ESP32-S3能否外接更多的天线来得到更多子载波信息？

MacChu0315-Espressif, 2024-04-02:
The singular column is the imaginary part, and the even column is the real part;
Phase information can be obtained;
At the same time, CSI data can only be obtained from frames collected from one antenna.

Worange12, 2024-04-02:
> The singular column is the imaginary part, and the even column is the real part; Phase information can be obtained; At the same time, CSI data can only be obtained from frames collected from one antenna.

感谢您解惑。这是否意味着我无法得到相位差信息了，因为相位差似乎需要至少两根天线，而esp32-s3只有板载天线。

MacChu0315-Espressif, 2024-04-02:
If you can get the phase of the sending device, that's OK. You can't get a phase difference on a single development board.

Worange12, 2024-04-02:
> If you can get the phase of the sending device, that's OK. You can't get a phase difference on a single development board.

好的，非常感谢

Worange12, 2024-04-02:
> If you can get the phase of the sending device, that's OK. You can't get a phase difference on a single development board.

e......最后一个问题，抱歉打扰您这么久。如果是直接连路由器的话我可以怎么做来获取发送方的相位信息呢。如果使用两个esp32是不是就可以得到发送方和接收方的相位信息了。