espressif/esp-csi #171: When I use esp-idf v5.1.3，then flash and monitor .I can't get console. The main_task is return。

qianxiaojun1, 2024-04-10:
xiaojunqian@xiaojun:/mnt/s1/esp_project/esp-csi/examples/esp-radar/console_test$ idf.py -p /dev/ttyACM0 monitor
Executing action: monitor
Running idf_monitor in directory /mnt/s1/esp_project/esp-csi/examples/esp-radar/console_test
Executing "/home/xiaojunqian/.espressif/python_env/idf5.1_py3.8_env/bin/python /home/xiaojunqian/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 2000000 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /mnt/s1/esp_project/esp-csi/examples/esp-radar/console_test/build/console_test.elf -m '/home/xiaojunqian/.espressif/python_env/idf5.1_py3.8_env/bin/python' '/home/xiaojunqian/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 2000000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037af36
0x4037af36: esp_cpu_wait_for_intr at /home/xiaojunqian/esp/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xdb0
load:0x403cc700,len:0x3084
entry 0x403c993c
I (26) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (27) boot: compile time Apr 10 2024 16:13:32
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) qio_mode: Enabling default flash chip QIO
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (28) boot.esp32s3: SPI Mode       : QIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (29) boot: Partition Table:
I (29) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (30) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (30) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (31) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (31) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (31) boot:  5 ota_0            OTA app          00 10 00020000 001ca000
I (32) boot: End of partition table
I (32) esp_image: segment 0: paddr=00020020 vaddr=3c0a0020 size=250cch (151756) map
I (56) esp_image: segment 1: paddr=000450f4 vaddr=3fc98200 size=0459ch ( 17820) load
I (59) esp_image: segment 2: paddr=00049698 vaddr=40374000 size=06980h ( 27008) load
I (65) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=960a4h (614564) map
I (159) esp_image: segment 4: paddr=000e60cc vaddr=4037a980 size=0d788h ( 55176) load
I (178) boot: Loaded app from partition at offset 0x20000
I (179) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (179) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x4037560c
0x4037560c: call_start_cpu1 at /home/xiaojunqian/esp/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (189) cpu_start: Pro cpu start user code
I (189) cpu_start: cpu freq: 240000000 Hz
I (189) cpu_start: Application information:
I (189) cpu_start: Project name:     console_test
I (189) cpu_start: App version:      2d79acf-dirty
I (190) cpu_start: Compile time:     Apr 10 2024 16:13:28
I (190) cpu_start: ELF file SHA256:  aab1120b9c1d4a0e...
I (190) cpu_start: ESP-IDF:          v5.1.3
I (191) cpu_start: Min chip rev:     v0.0
I (191) cpu_start: Max chip rev:     v0.99 
I (191) cpu_start: Chip rev:         v0.2
I (191) heap_init: Initializing. RAM available for dynamic allocation:
I (192) heap_init: At 3FCA0B10 len 00048C00 (291 KiB): DRAM
I (192) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (193) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (193) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (194) spi_flash: detected chip: gd
I (194) spi_flash: flash io: qio
W (194) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (195) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (195) sleep: Configure to isolate all GPIO pins in sleep state
I (196) sleep: Enable automatic switching of GPIO sleep configuration
I (196) app_start: Starting scheduler on CPU0
I (196) app_start: Starting scheduler on CPU1
I (196) main_task: Started on CPU0
I (197) main_task: Calling app_main()

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
csi> I (707) pp: pp rom version: e7ae62f
I (708) net80211: net80211 rom version: e7ae62f
I (710) wifi:wifi driver task: 3fcad008, prio:23, stack:6656, core=0
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
I (713) wifi:wifi firmware version: 0016c4d
I (713) wifi:wifi certification version: v7.0
I (714) wifi:config NVS flash: enabled
I (714) wifi:config nano formating: disabled
I (715) wifi:Init data frame dynamic rx buffer num: 32
I (716) wifi:Init static rx mgmt buffer num: 5
I (716) wifi:Init management short buffer num: 32
I (717) wifi:Init dynamic tx buffer num: 32
I (717) wifi:Init static tx FG buffer num: 2
I (718) wifi:Init static rx buffer size: 2212
I (718) wifi:Init static rx buffer num: 10
I (719) wifi:Init dynamic rx buffer num: 32
I (720) wifi_init: tcpip mbox: 32
I (720) wifi_init: udp mbox: 6
I (721) wifi_init: tcp mbox: 6
I (721) wifi_init: tcp tx win: 5760
I (722) wifi_init: tcp rx win: 5760
I (722) wifi_init: tcp mss: 1440
I (723) wifi_init: WiFi IRAM OP enabled
I (723) wifi_init: WiFi RX IRAM OP enabled
I (724) phy_init: phy_version 640,cd64a1a,Jan 24 2024,17:28:12
I (761) wifi:mode : null
I (762) wifi:Set ps type: 0, coexist: 0

I (763) wifi:ic_enable_sniffer
I (763) esp_radar: CSI_RADAR Version: 0.1.0, Compile time: Jun  6 2023 - 14:20:50
I (765) esp_radar: csi_recv_interval: 10, csi_handle_time: 250, csi_handle_window: 50, csi_handle_buffer: 60
I (766) main_task: Returned from app_main()

qianxiaojun1, 2024-04-10:
if it is correct，would I get like “csi>” page ?

qianxiaojun1, 2024-04-10:
by the way,my device is esp32s3

MacChu0315-Espressif, 2024-04-12:
#151 may help.