espressif/esp-csi #172: esp-csi/examples/esp-radar/console_test not working

rav2n-dev, 2024-04-11:
Following the instructions doesn't give any waveform in the gui.

Please help as soon as possible, it is very urgent (for a humanitarian cause).

MacChu0315-Espressif, 2024-04-12:
Does baud rate changing according #151 help?

rav2n-dev, 2024-04-12:
> Does baud rate changing according to #151 help?

Made sure the baud rate was matching.
Still doesn't display any detail (waveform or logs or anything) unfortunately. It's all blank..

MacChu0315-Espressif, 2024-04-22:
To better diagnose the issue, please monitor the ESP device using the command `idf.py monitor` and then enter the following two commands:

1. `radar --csi_output_type LLFT --csi_output_format base64`
2. `wifi_config --ssid "ROUTER_NAME" --password ROUTER_PASSWORD`

Additionally, could you please provide detailed descriptions of your testing environment? Include the type of development kit you are using, the version of the IDF, and any other relevant information. This information will greatly assist us in diagnosing and addressing your issue more effectively.

q936505350, 2024-05-11:
[image]
@MacChu0315-Espressif I have the same issue，and this is  the situation after I ran monitor……And, if I flash the "get-started\csi_recv" code, I can receive CSI data normally, but then I won't be able to use esp_csi_tools. I sincerely ask you how to solve this problem, please.

MacChu0315, 2024-05-20:
Hi @q936505350: Firstly, regarding the image you provided, it's possible that the issue stems from an incorrect baud rate setting and we can't get enough info starting from the secondary bootloader. To assist you more effectively, could you please provide more detailed information? Secondly, the esp_csi_tool is utilized in conjunction with the console_test program. Lastly, based on the information provided, your issue does not seem to be closely related to the existing discussions. It is recommended that you initiate a new issue for further discussion.

q936505350, 2024-05-20:
Thank you for your reply, I have resolved the issue.

936505350
***@***.***

rav2n-dev, 2024-06-13:
> radar --csi_output_type LLFT --csi_output_format base64
> wifi_config --ssid "ROUTER_NAME" --password ROUTER_PASSWORD

Trying these out provides csi data logs and the controller seems to be running the flashed code (the sound it generates is not so low).

But the GUI doesn't display anything.. 

IDF vdersion: 5.0.2
CSI_RADAR Version: 0.2.0

OS: linux 

Examples to use:
https://github.com/espressif/esp-csi/blob/master/examples/get-started/csi_recv_router/README.md

and visualize using:
https://github.com/espressif/esp-csi/tree/master/examples/esp-radar/console_test

tried flashing csi_recv_router then console_test/tool/esp_csi_tool.py
also tried flashing console_test then console_test/tool/esp_csi_tool.py

basically followed the READMEs in csi_recv_router (as I only have one esp32) and console_test.

Also set the matching right baud rate.

MacChu0315-Espressif, 2024-07-01:
Hi @rav2n-dev, can you provide the output from running the following two lines of code?

> To better diagnose the issue, please monitor the ESP device using the command `idf.py monitor` and then enter the following two commands:
>
> 1. `radar --csi_output_type LLFT --csi_output_format [...]