espressif/esp-csi #173: 关于get-started示例的一些问题

TS-Sun-P, 2024-04-19:
有以下几个使用问题想进行请教：
[image]
1.为何在相同条件（secondary channel : below, HT, 40 MHz, non STBC, v, HT-LFT: 0~63, -64~-1, 384）下运行时，有时收到的CSI数据是256bytes，有时是384bytes，这是由于什么原因导致的呢？还是本身存在的固有缺陷呢？是否存在能固定收发数据的方法呢？另外相较384bytes数据，收到256bytes时，是少了哪一部分数据呢？
2.关于具体的子载波顺序排列是按照此LLTF：0-63，HT-LFT: 0~63, -64~-1顺序排列的吗？
3.可否就LLTF和HT-LFT给出一些更加详细的官方说明和解释，解释具体的区别和原理以及二者之间存在的关系吗？
4.可否对这一示例中的发包频率进行解释。我们发现烧录代码中发送频率为100，但实际处理解析后，一分钟发包量为4700左右，这二者并不对应，可否对此进行解释。
[image]

5.关于上一问题中的发包频率，esp32最多可以支持多少呢？具体是否可以给出调整示例
期待并感谢您的回复。

MacChu0315, 2024-04-23:
1. Due to adverse air conditions, the transmitting end proactively adjusted the transmission protocol, as stipulated by the protocol. To address this issue, we will subsequently use null_data to obtain CSI. The specific protocol standard that was adjusted can be determined by comparing the differences between these two scenarios.

2. Yes.

3. This point can be referenced in the protocol itself or in related materials, such as [802.11ac: A Survival Guide](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/index.html?_gl=1*3tyahb*_ga*MTM3NjEwNjg1OC4xNjkyNzczODEz*_ga_092EL089CH*MTY5Mjc3MzgxMi4xLjEuMTY5Mjc3NDA4My44LjAuMA..).

4. Packet loss during aerial packet transmission is a normal occurrence.

5. It can achieve a packet transmission frequency of about 2~3K. You can directly modify the sending frequency, transmission rate, or disable ampdu settings based on the example program.

MacChu0315-Espressif, 2024-04-23:
1. Due to adverse air conditions, the transmitting end proactively adjusted the transmission protocol, as stipulated by the protocol. To address this issue, we will subsequently use null_data to obtain CSI. The specific protocol standard that was adjusted can be determined by comparing the differences between these two scenarios.

2. Yes.

3. This point can be referenced in the protocol itself or in related materials, such as [802.11ac: A Survival Guide](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/index.html?_gl=1*3tyahb*_ga*MTM3NjEwNjg1OC4xNjkyNzczODEz*_ga_092EL089CH*MTY5Mjc3MzgxMi4xLjEuMTY5Mjc3NDA4My44LjAuMA..).

4. Packet loss during aerial packet transmission is a normal occurrence.

5. It can achieve a packet transmission frequency of about 2~3K. You can directly modify the sending frequency, transmission rate, or disable ampdu settings based on the example program.

TS-Sun-P, 2024-04-23:
非常感谢您的回答，关于第四点提问，可否请您再详细解释一下：
在具体使用这一example的实验中，我们使用其并通过手动计时采集约1min的数据（1min时手动关闭数据接收），但从收集到的csv文件（如下）中可以看到
[image]
[image]
根据序号计算下来，对应实际包数为4542个，我们理解其为发送端1min内发送的数据
根据数据条数算下来，对应的实际包数也为4542个，我们理解其为接收端实际收到的数据
由于之前数据采集中出现过两者有小差值，即序号计算下包数比数据条数多几个，我们理解为这是出现丢包，这样表明丢包率很低；因此在上面图片呈现的这一过程中，我们理解为其没有出现丢包的情况，那这代表其1min实际发包数就4542个左右，和理论发包率为100的情况对应1min发包数6000差距很大，是我理解对于丢包、收包有误吗？或是我们操作有误？还是实际就是不匹配的问题呢？亦或是数据开始收发写入csv文件存在较大延迟呢？若存在延迟，可以请您给出一个较为合适的参考值吗？
可以请您再回答一下吗？
期待并感谢您的恢复！

MacChu0315-Espressif, 2024-04-24:
The two values you mentioned both represent the sequence numbers of packets received by the receiver. In other words, "sequence number" and "number of data items" refer to the same information. To confirm how many packets have been sent during a specified time period, you can add a print statement to log the number of packets sent by the transmitter. Then, compare this value with the number of packets received by the receiver.