espressif/esp-csi #174: Receive no CSI info after changing the receiver's MAC address

liuwuhaoo, 2024-04-29:
I am running the "esp-csi/examples/get-started" example and the problem is that I can receive CSI data as expected. The problem is that if I change the MAC address from {0x1a, 0x00, 0x00, 0x00, 0x00, 0x00} to, for example, {0x2a, 0x00, 0x00, 0x00, 0x00, 0x00} (in line 25 of examples/get-started/csi_recv/main/app_main.c), then I receive nothing. I did some tests and it seems that the receiver can only receive CSI data if its MAC address is the same as the MAC address of the sender (set by esp_wifi_set_mac). I'm confused, isn't that something wrong, I mean two devices have the same MAC address?

Thank you very much.

MacChu0315-Espressif, 2024-04-29:
Although different ESP devices have different MAC addresses, in this demonstration, we aim to use the MAC address as a filter to more clearly identify the sender and receiver. Of course, you can also filter wireless packets from your desired sender using other methods.

liuwuhaoo, 2024-04-29:
Thanks for your reply.

So why the receiver gets no CSI information when I change the receiver's MAC address? I see in this demonstration, the packages are broadcasted (0xff, 0xff, 0xff, 0xff, 0xff). The receiver's MAC address should not change anything, right?

MacChu0315-Espressif, 2024-04-29:
No need to change the receiver's MAC but to change the filter MAC (CONFIG_CSI_SEND_MAC) in csi_recv code. Give it a try :D

liuwuhaoo, 2024-05-14:
Thanks for the information. Yes, I missed the filter line in the code.

closed 2024-05-14 by liuwuhaoo