espressif/esp-csi #175: Relate CSI information between devices

liuwuhaoo, 2024-05-14:
Now I am trying to do key generation from CSI data. I am collecting CSI data by sending probe packets. A sender sends a probe packet with an identifier and the receiver responds with the same identifier. In this case, both should have a matching CSI record. The problem is that the CSI data might get lost. Sometimes the sender gets more CSI data than the receiver or vice versa. 

How can I match the CSI data from each side? One simple way would be to also store the identifier, but it seems packet data is not accessible in CSI received callback function.

MacChu0315, 2024-05-20:
The functionality you mentioned is already achievable. On the master IDF, under components/esp_wifi/include/local/esp_wifi_types_native.h, the wifi_csi_info_t already offers a more comprehensive set of data. You can adapt this on your own to implement the desired features.

MacChu0315-Espressif, 2024-05-20:
The functionality you mentioned is already achievable. On the master IDF, under components/esp_wifi/include/local/esp_wifi_types_native.h, the wifi_csi_info_t already offers a more comprehensive set of data. You can adapt this on your own to implement the desired features.

liuwuhaoo, 2024-05-23:
Thanks for your answer.
In this wifi_csi_info_t structure, I can get the source MAC address and the destination MAC address, so I can match who is the sender and who is the recipient. This is good. However, I cannot know which packet this CSI is associated with.

```
typedef struct {
    wifi_pkt_rx_ctrl_t rx_ctrl;/**< received packet radio metadata header of the CSI data */
    uint8_t mac[6];            /**< source MAC address of the CSI data */
    uint8_t dmac[6];           /**< destination MAC address of the CSI data */
    bool first_word_invalid;   /**< first four bytes of the CSI data is invalid or not */
    int8_t *buf;               /**< buffer of CSI data */
    uint16_t len;              /**< length of CSI data */
} wifi_csi_info_t;
```

I want to know the corresponding packet data. For example, provide a data parameter in csi callback function. Namely from `typedef void (* wifi_csi_cb_t)(void *ctx, wifi_csi_info_t *data);` to `typedef void (* wifi_csi_cb_t)(void *ctx, wifi_csi_info_t *data, const uint8_t *packet_data);`.

MacChu0315-Espressif, 2024-05-24:
There may have been some misunderstandings in the communication. What I meant to convey is that the `wifi_csi_info_t` structure on the master branch of IDF already provides more information, as shown below:

```c
typedef struct wifi_csi_info_t {
    wifi_pkt_rx_ctrl_t rx_ctrl; /**< received packet radio metadata header of the CSI data */
    uint8_t mac[6];             /**< source MAC address of the CSI data */
    uint8_t dmac[6];            /**< destination MAC address of the CSI data */
    bool first_word_invalid;    /**< first four bytes of the CSI data are invalid or not; true indicates the first four bytes are invalid due to hardware limitations */
    int8_t *buf;                /**< valid buffer of CSI data */
    uint16_t len;               /**< valid length of CSI data */
    uint8_t *hdr;               /**< header of the wifi packet */
    uint8_t *payload;           /**< payload of the wifi packet */
    uint16_t payload_len;       /**< payload length of the wifi packet */
    uint16_t rx_seq;            /**< rx sequence number of the wifi packet */
} wifi_csi_info_t;
```

After adapting this to our project, it allows you to match the transmitter and receiver using the payload of the wireless packet.

liuwuhaoo, 2024-05-24:
Problem solved. Thank you.

closed 2024-05-24 by liuwuhaoo