espressif/esp-csi #177: esp-radar/console_test

xlwan1132, 2024-06-01:
我使用的是ESP-IDF Release v5.1
cd esp-csi/examples/esp-radar/console_test
idf.py set-target esp32s3时报错：

    CMake Error at D:/WIFI/esp-idf-v5.1/tools/cmake/build.cmake:540 (message):
  ERROR: Because no versions of esp-radar match >=0.0.0,<0.2.0 ||
  >0.2.0,<1.0.0

   and esp-radar (0.2.0) depends on espressif/cmake_utilities (0.*), esp-radar (>=0.0.0,<1.0.0) requires espressif/cmake_utilities (0.*).

  So, because no versions of espressif/cmake_utilities match 0.*

   and project depends on esp-radar (0.*), version solving failed.

Call Stack (most recent call first):
  D:/WIFI/esp-idf-v5.1/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:10 (project)
请问怎么解决

MacChu0315-Espressif, 2024-06-03:
This may be due to your repository not being fully synchronized. You can try running the git pull command in the project directory and then attempt again. If the issue persists, please provide the specific contents of the components/esp-radar/idf_component.yaml and console_test/managed_components/espressif__cmake_utilities/idf_component.yaml files.

tempecks, 2024-07-16:
I encountered the same issue.
After commanding `git pull` in the directry of console_test, showing only "Already up to date"
Please advise me how to solve it.

### The specific contents of the componets/esp-radar/idf_component.yaml
version: "0.2.0"
description: This package provides algorithmic functionality for ESP-CSI human movement and presence detection for easy integration into products.
url: https://github.com/espressif/esp-csi/tree/master/esp-radar
issues: https://github.com/espressif/esp-csi/issues
targets:
  - esp32
  - esp32s3
  - esp32c3
dependencies:
  idf: ">=4.4.1"
  cmake_utilities: "0.*"
examples:
  - path: ../../examples/esp-radar/connect_rainmaker
  - path: ../../examples/esp-radar/console_test
  
  
  
### The specific contents of the examples/esp-radar/console_test/main/idf_component.yaml
## IDF Component Manager Manifest File
dependencies:
  idf: ">=4.4.1"

  esp-radar:
    version: "0.*"
    override_path: "../../../../components/esp-radar"

  espressif/ws2812_led:
    path: components/ws2812_led
    git: https://github.com/espressif/esp-rainmaker.git
    version: "ff0dd329020a52ae8cf2fb4f6800d9f8e9cc9eeb"

MacChu0315-Espressif, 2024-07-16:
Can you provide the completed output log when running commands `idf.py python-clean; idf.py fullclean; rm -rf build; idf.py set-target esp32c3; idf.py flash`? And please make sure your have run the commands `git pull; git submodule update --init --recursive; 
 ./install.sh; . ./export.sh;` at the root dirctory of esp-idf.