espressif/esp-csi #179: Will esp-csi support esp32c6 target?

sensaurabh, 2024-06-05:


MacChu0315-Espressif, 2024-06-06:
Yes, will do.

TryTwoTop, 2024-07-04:
@MacChu0315-Espressif 
Hello

I have successfully collected 512 subcarriers of HE20 with the ESP32-C6 official development board.
I would like to know the index of null carrier and pilot carrier in the subcarrier.
Will you update the information about HE20 subcarriers of ESP32-C6 in ESP-IDF Programming Guide in the future?

TryTwoTop, 2024-07-25:
@MacChu0315-Espressif

I look forward to hearing from you.
And for questions like this, should I submit a new issue?

fathanpranaya, 2024-08-29:
hi @TryTwoTop may I know how to enable this on esp32-c6 and which idf version you use?

thank you

TryTwoTop, 2024-08-29:
Hi @fathanpranaya 

ESP32-C6 is supported starting with ESP-IDF v5.1, so I used v5.1. [Link-1]

I have modified the csi_recv, csi_send and tools/csi_data_read_parse.py examples to collect and store CSI data from the ESP32-C6.

Please refer to the ESP-IDF Programming Guide and modify the code as it is different from the existing code currently uploaded. [Link-2]

Link-1: https://github.com/espressif/esp-idf?tab=readme-ov-file#esp-idf-release-and-soc-compatibility
Link-2: https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32c6/index.html

MacChu0315-Espressif, 2024-08-29:
Thanks to you @TryTwoTop. And @fathanpranaya you can make a try following TRY's guides. We have now intergrated csi example of csi_send and csi_recv for C6 and other 32 chips in our internal branch. And these codes will be disclosed in near furture.

MacChu0315-Espressif, 2024-08-29:
> I look forward to hearing from you. And for questions like this, should I submit a new issue?

BTW, we are now in progress of improving the part of idf-programing guide relates to [CSI](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/wifi.html#id61). For we found that the insufficient of explanition of protocol (including data parsing, subcarrier identification and so on ...) bothers people a lot. We will gradually improve the writings of 32/c3/s3, and then C5/C6. Thank you for your interest in this project.