espressif/esp-csi #183: Questions about supporting and enabling beamforming

TryTwoTop, 2024-07-01:
There are two questions in this issue.

You can answer them one by one, or we'll create separate issues if requested.

1. Does the ESP32-S3 support beamforming?
(I can't find anything about beamforming in the ESP32-S3's datasheet and esp-idf GitHub repo)
    
2. Does the ESP32-C6 have beamforming automatically enabled only when communicating with 802.11ax?
(You can find 802.11ax beamformee support in the Wi-Fi Feature of the ESP32-C6 Datasheet. )

MacChu0315-Espressif, 2024-07-01:
1. Not supported
2. This can be set by `wifi_csi_acquire_config_t` in `esp_wifi_he_types.h`. By the way, the current code cannot be directly run on the c6 chip.

fpgasdr, 2025-02-20:
Hi @MacChu0315-Espressif ,what devices support beamforming?.

NingYuxiang, 2025-02-21:
Currently, the ESP32-C5, ESP32-C6, and ESP32-C61 support beamforming functionality.

closed 2025-10-15 by TryTwoTop