espressif/esp-csi #184: ESP32-C6-DevKit with an external antenna connecter

TryTwoTop, 2024-07-02:
The issue below says that it will support ESP32-C6, but the official ESP32-C6-based dev boards that currently exist are with built-in antennas.

Are there any plans to release a dev board with an external antenna?

If not, will I have to build my own dev board with a module with an external antenna connector?

MacChu0315-Espressif, 2024-07-03:
Which issue are you referring to?
As far as I know, there isn't one. And, what functionality are you looking to implement?

TryTwoTop, 2024-07-03:
@MacChu0315-Espressif 

Sorry about that. I forgot to include the link
[GitHub Issue: https://github.com/espressif/esp-csi/issues/179]

It says that using an external antenna works better than using a PCB antenna, so I asked if there are any plans to release a dev board with an external antenna.
*reference link
https://github.com/espressif/esp-csi?tab=readme-ov-file#5-note

MacChu0315-Espressif, 2024-07-03:
I apologize, but we currently do not have plans for this.
If you wish to implement such a feature, there are several ways to achieve it:
    1. Design your own based on the 1u module.
    2. Use a development board with a PCB antenna, remove the existing module and replace it with a 1u module.

closed 2024-08-13 by TryTwoTop