espressif/esp-csi #185: Call for hep about automatic gain control (AGC)

zaoanhh, 2024-07-03:
I am very grateful to the author for developing such a clever tool to use, but I have a question in the use of which I would like to get your answer. It is noted that the csi data provided by the ESP32 chip does not contain automatic gain control (AGC) information, which is different from the CSI TOOl. I would like to ask you, in this case, how should I get the channel frequency response of the actual signal strength?

MacChu0315-Espressif, 2024-07-03:
在接收到 CSI 信息后，对 `wifi_csi_info_t *info` 进行格式转换后即可获得 agc 信息：

```
wifi_pkt_rx_ctrl_phy_t *phy_info = (wifi_pkt_rx_ctrl_phy_t *)info;
ESP_LOGI(TAG, "agc = %d and fft = %d", phy_info->agc_gain, phy_info->fft_gain);
```

其中，`wifi_pkt_rx_ctrl_phy_t` 如下所示：

```c
typedef struct {
    signed rssi:8;                /**< Received Signal Strength Indicator(RSSI) of packet. unit: dBm */
    unsigned rate:5;              /**< PHY rate encoding of the packet. Only valid for non HT(11bg) packet */
    unsigned :1;                  /**< reserved */
    unsigned sig_mode:2;          /**< 0: non HT(11bg) packet; 1: HT(11n) packet; 3: VHT(11ac) packet */
    unsigned :16;                 /**< reserved */
    unsigned mcs:7;               /**< Modulation Coding Scheme. If is HT(11n) packet, shows the modulation, range from 0 to 76(MSC0 ~ MCS76) */
    unsigned cwb:1;               /**< Channel Bandwidth of the packet. 0: 20MHz; 1: 40MHz */
    unsigned :16;                 /**< reserved */
    unsigned smoothing:1;         /**< reserved */
    unsigned not_sounding:1;      /**< reserved */
    unsigned :1;                  /**< reserved */
    unsigned aggregation:1;       /**< Aggregation. 0: MPDU packet; 1: AMPDU packet */
    unsigned stbc:2;              /**< Space Time Block Code(STBC). 0: non STBC packet; 1: STBC packet */
    unsigned fec_coding:1;        /**< Flag is set for 11n packets which are LDPC */
    unsigned sgi:1;               /**< Short Guide Interval(SGI). 0: Long GI; 1: Short GI */
#if CONFIG_IDF_TARGET_ESP32
    signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
#elif CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
    unsigned :8;                  /**< reserved */
#endif
    unsigned ampdu_cnt:8;         /**< ampdu cnt */
    unsigned channel:4;           /**< primary channel on which this packet is received */
    unsigned secondary_channel:4; /**< secondary channel on which this packet is received. 0: none; 1: above; 2: below */
    unsigned :8;                  /**< reserved */
    unsigned timestamp:32;        /**< timestamp. The local time when this packet is received. It is precise only if modem sleep or light sleep is not enabled. unit: microsecond */
    unsigned :32;                 /**< reserved */
#if CONFIG_IDF_TARGET_ESP32S2
    unsigned :32;                 /**< reserved */
#elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
    signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
    unsigned :8;                 /**< reserved */
    unsigned fft_gain: 8;
    unsigned agc_gain: 8;
    unsigned :32;                 /**< reserved */
#endif
    unsigned :31;                 /**< reserved */
    unsigned ant:1;               /**< antenna number from which this packet is received. 0: WiFi antenna 0; 1: WiFi antenna 1 */
#if CONFIG_IDF_TARGET_ESP32S2
    signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
    unsigned :24;                 /**< reserved */
#elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
    unsigned :32;                 /**< reserved */
    unsigned :32;                 /**< reserved */
    unsigned :32;                 /**< reserved */
#endif
    unsigned sig_len:12;          /**< length of packet including Frame Check Sequence(FCS) */
    unsigned :12;                 /**< reserved */
    unsigned rx_state:8;          /**< state of the packet. 0: no error; others: error numbers which are not public */
} wifi_pkt_rx_ctrl_phy_t;
```

bitailab, 2024-10-15:
[image]
请教下，只有S3和 C3 能获取到 agc_gain 吗？

bitailab, 2024-10-15:
> After receiving the CSI information, agc information can be obtained by converting the format of 'wifi_csi_info_t *info' :
> 
> ```
> wifi_pkt_rx_ctrl_phy_t *phy_info = (wifi_pkt_rx_ctrl_phy_t *)info;
> ESP_LOGI(TAG, "agc = %d and fft = %d", phy_info->agc_gain, phy_info->fft_gain);
> ```
> 
> Where, 'wifi_pkt_rx_ctrl_phy_t' is as follows:
> 
> ```c
> typedef struct {
>     signed rssi:8;                /**< Received Signal Strength Indicator(RSSI) of packet. unit: dBm */
>     unsigned rate:5;              /**< PHY rate encoding of the packet. Only valid for non HT(11bg) packet */
>     unsigned :1;                  /**< reserved */
>     unsigned sig_mode:2;          /**< 0: non HT(11bg) packet; 1: HT(11n) packet; 3: VHT(11ac) packet */
>     unsigned :16;                 /**< reserved */
>     unsigned mcs:7;               /**< Modulation Coding Scheme. If is HT(11n) packet, shows the modulation, range from 0 to 76(MSC0 ~ MCS76) */
>     unsigned cwb:1;               /**< Channel Bandwidth of the packet. 0: 20MHz; 1: 40MHz */
>     unsigned :16;                 /**< reserved */
>     unsigned smoothing:1;         /**< reserved */
>     unsigned not_sounding:1;      /**< reserved */
>     unsigned :1;                  /**< reserved */
>     unsigned aggregation:1;       /**< Aggregation. 0: MPDU packet; 1: AMPDU packet */
>     unsigned stbc:2;              /**< Space Time Block Code(STBC). 0: non STBC packet; 1: STBC packet */
>     unsigned fec_coding:1;        /**< Flag is set for 11n packets which are LDPC */
>     unsigned sgi:1;               /**< Short Guide Interval(SGI). 0: Long GI; 1: Short GI */
> #if CONFIG_IDF_TARGET_ESP32
>     signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
> #elif CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
>     unsigned :8;                  /**< reserved */
> #endif
>     unsigned ampdu_cnt:8;         /**< ampdu cnt */
>     unsigned channel:4;           /**< primary channel on which this packet is received */
>     unsigned secondary_channel:4; /**< secondary channel on which this packet is received. 0: none; 1: above; 2: below */
>     unsigned :8;                  /**< reserved */
>     unsigned timestamp:32;        /**< timestamp. The local time when this packet is received. It is precise only if modem sleep or light sleep is not enabled. unit: microsecond */
>     unsigned :32;                 /**< reserved */
> #if CONFIG_IDF_TARGET_ESP32S2
>     unsigned :32;                 /**< reserved */
> #elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
>     signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
>     unsigned :8;                 /**< reserved */
>     unsigned fft_gain: 8;
>     unsigned agc_gain: 8;
>     unsigned :32;                 /**< reserved */
> #endif
>     unsigned :31;                 /**< reserved */
>     unsigned ant:1;               /**< antenna number from which this packet is received. 0: WiFi antenna 0; 1: WiFi antenna 1 */
> #if CONFIG_IDF_TARGET_ESP32S2
>     signed noise_floor:8;         /**< noise floor of Radio Frequency Module(RF). unit: dBm*/
>     unsigned :24;                 /**< reserved */
> #elif CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3
>     unsigned :32;                 /**< reserved */
>     unsigned :32;                 /**< reserved */
>     unsigned :32;                 /**< reserved */
> #endif
>     unsigned sig_len:12;          /**< length of packet including Frame Check Sequence(FCS) */
>     unsigned :12;                 /**< reserved */
>     unsigned rx_state:8;          /**< state of the packet. 0: no error; others: error numbers which are not public */
> } wifi_pkt_rx_ctrl_phy_t;
> ```
[image]
请教下，只有S3和 C3 能获取到 agc_gain 吗？