espressif/esp-csi #186: csi_data-read_parse error

xlwan1132, 2024-07-04:
``
if len(csi_data) != len(DATA_COLUMNS_NAMES):
    print("**element number is not equal**")
    continue

try:
    csi_raw_data = json.loads(csi_data[-1])
except json.JSONDecodeError:
    print(f"**data is incomplete**")
    continue

if len(csi_raw_data) != 128 and len(csi_raw_data) != 256 and len(csi_raw_data) != 384:
    print(f"**element number is not equal:** {len(csi_raw_data)}")
    continue
``
[image]

What is the reason for these errors when I use csi_data-read_parse for data collection？

suhipek, 2024-07-04:
Thank you for reporting this issue. It seems like the CSI data is corrupted when transferring on the serial.

To help us investigate:

1. Which development board / serial-to-USB chip are you using?
2. Have you checked the serial output with `idf.py monitor`?
3. Do you see corrupted packets in the serial output?

Please provide a dump of your serial output if possible. Any additional details on how to reproduce the error would be helpful.

xlwan1132, 2024-07-05:
1.I am using the ESP32 S3 development board
2.sorry,I don't know if you want me to monitor sending or receiving COM？

suhipek, 2024-07-05:
> 1.I am using the ESP32 S3 development board
> 2.sorry,I don't know if you want me to monitor sending or receiving COM？

You should monitor the receiver - that's where the [csi_data_read_parse.py](https://github.com/espressif/esp-csi/blob/master/examples/get-started/tools/csi_data_read_parse.py) script grabs the CSI data from. 

I've added some print statements to show the packet content when it spots a corrupted packet. With those print statements you can check corrupted packets directly through console. Check out the updated csi_data_read_parse.py here: https://gist.github.com/suhipek/4f65fc797844b24ded20c8be0ab45162

BTW, have you changed any parameters in the sender, receiver, or parser programs?

xlwan1132, 2024-07-12:
[error.txt](https://github.com/user-attachments/files/16190797/error.txt)

xlwan1132, 2024-07-19:
@suhipek

MacChu0315-Espressif, 2024-07-19:
This is a strange phenomenon. If you look at the log you gave, there are many pieces of data that are broken. 
Have you ever tried to change the code? At the same time, do you have any external equipment attached to your development board, and have you tried to replace a development board and cable for comparison testing to eliminate hardware interference?

xlwan1132, 2024-07-22:
1.I cloned the esp-csi code again, but this error still occurs
2.There are no other devices connected to my development board
3.I used another set of development boards for sending and receiving，but this error still occurs