espressif/esp-csi #188: setup for esp-csi console_test

tempecks, 2024-07-10:
I encountered the error witten below when setting up for esp-csi console_test.
How can I solve this problem? Please help me, very beginner.

PS C:\esp-csi\examples\esp-radar\console_test> idf.py set-target esp32-c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory 'C:\esp-csi\examples\esp-radar\console_test\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.

MacChu0315-Espressif, 2024-07-11:
performing commands below:  
`idf.py python-clean; idf.py fullclean; rm -rf build;` 
and try to rerun the commands following the guide of [Readme of Console_test](https://github.com/espressif/esp-csi/tree/master/examples/esp-radar/console_test)  may help.

Besides, [Getting started of ESP-IDF](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/get-started/index.html) may help too.

MacChu0315-Espressif, 2024-07-16:
Can you provide the completed output log when running commands `idf.py python-clean; idf.py fullclean; rm -rf build; idf.py set-target esp32c3; idf.py flash`? And please make sure your have run the commands `git pull; git submodule update --init --recursive; 
 ./install.sh; . ./export.sh;` at the root dirctory of esp-idf.

Yes, esp-radar can work on ESP32-C3.

tempecks, 2024-07-16:
Here are the completed output log when running commands  `git pull; git submodule update --init --recursive;  ./install.sh; . ./export.sh`; at the dirctory of esp-idf-v5.0.4;
 
PS C:\Espressif\frameworks\esp-idf-v5.0.4> git pull
error: cannot spawn gpg: No such file or directory
Already up to date.
PS C:\Espressif\frameworks\esp-idf-v5.0.4> git submodule update --init --recursive
PS C:\Espressif\frameworks\esp-idf-v5.0.4> ./install.sh
PS C:\Espressif\frameworks\esp-idf-v5.0.4>
[main 2024-07-16T11:14:29.152Z] update#setState idle
[main 2024-07-16T11:14:59.158Z] update#setState checking for updates
PS C:\Espressif\frameworks\esp-idf-v5.0.4> tate idle
PS C:\Espressif\frameworks\esp-idf-v5.0.4> . ./export.sh

After commanding `./install.sh; . ./export.sh`, VS CODE popped up, but nothing happen except for the out put written above.

MacChu0315-Espressif, 2024-07-17:
Seems your development environment is' win+PowerShell '? I recommend that you follow the document: [how to use the WSL + VSCode build ESP32 development environment] (https://www.bilibili.com/read/cv7755144/) set up your development environment on Windows. Looking forward to your feedback.

tempecks, 2024-07-21:
Thank you for your comment.
Following the [document ](https://www.bilibili.com/read/cv7755144/), I encountered the situation witten below. Please how should I solve this?

-----------------------------------------------

******@********:~/esp-idf$ sudo apt-get install python3-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-venv is already the newest version (3.6.7-1~18.04).
The following packages were automatically installed and are no longer required:
  efibootmgr libefiboot1 libefivar1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 29 not upgraded.
******@*********:~/esp-idf$ python3 -m venv /home/******/.espressif/python_env/idf5.4_py3.8_env
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/******/.espressif/python_env/idf5.4_py3.8_env/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']