espressif/esp-csi #189: to identify actions and evaluate accuracy by using console test

s129729, 2024-07-18:
I'd like to identify a specific action and evaluate accuracy by using console test.
So, I collected the CSI data, for example, as the target of jump, sended the data, and click the csi start button while jumping for evaluating accuracy.
But the recognition result shows only  none/someone or move/static, then I can't evaluate identifing the action.
Doesn't it show the result of jump or not when sending jump data? I'd like to know if it is working correctly or not.

MacChu0315-Espressif, 2024-07-19:
The present algorithm only provides the detection of whether an action occurs in the presence or absence of a person. The functionality you mentioned above is used to annotate actions to help you build your own dataset.

s129729, 2024-07-21:
Thank you for your reply.
I understand the functionality of "5.5 Collect CSI data for a specific action"
Still I cannot understand the fuctionality of "5.6 Use collected data to identify actions and evaluate accuracy".
Can't I use this for evaluating a specific action?

MacChu0315-Espressif, 2024-07-23:
Hello, I have confirmed that the function mentioned in section 5.6 has been removed, but the UI and readme have not been changed in time. Sorry for the inconvenience caused to you.

MacChu0315-Espressif, 2024-07-23:
We will fix this later, thank you!

tempecks, 2024-07-24:
Is it difficult to provide that function again? I'm very interested in it.

MacChu0315-Espressif, 2024-07-24:
This tool was originally used to internally test the detection algorithms used, but it was later found to be less useful than expected, and it was deprecated.