espressif/esp-csi #191: 大空间使用检测人体活动

daydayup33, 2024-07-24:
你好乐鑫工程师，我在小范围的封闭空间（20㎡）使用检测人体活动过取得很好的效果，现在我想使用在大范围的空间如70-100㎡，能否使用多套设备，多套设备的CSI会互相影响吗。

[image]
请问这里提到的外接天线有什么要求吗，官方是否有推荐的外接天线与安装使用教程。

MacChu0315-Espressif, 2024-07-24:
1. It is necessary to pay attention to not sending and receiving too frequently in the same channel to avoid channel pollution and not to have too many devices using the same channel.
2. For external antenna selection, as long as 50ohm input impedance, 2.4G antenna can be selected. It's recommended to consult the antenna company for specific selection suggestions.

daydayup33, 2024-07-24:
感谢你的及时回复，我是否可以理解为在一个环境中放置两套设备，使用不一样的channel，那么两套设备CSI数据互相不影响，放在两个角落可以分别检测人体活动。
[image]
是否指的是这个channel，取指范围是否有要求。
另外还有一个问题，这个只能是一对一的吗，能否在一个环境中，中间放发送设备烧录csi_send，两边接受设备烧录console_test，以此扩大检测人体活动的范围。

MacChu0315-Espressif, 2024-07-29:
Hi @daydayup33,
1. I do refer to this ‘channel', but it should be noted that when choosing channels that close to each other to send, they will still be interfered with each other, so you can try to choose channels such as 1, 6, 11.
2. The situation you described is feasible.

closed 2024-08-20 by daydayup33