espressif/esp-csi #199: esp_radar_train warning and csi caluculate fail error

ashashik, 2024-08-20:
Hi, 

I am trying to use 2 esp32 boards with the steps in console_test and I see the following warnings when I try to train the data in  very short time . Is this normal ?  

```
W (34724) esp_radar: The capacity of subcarrier recording during calibration is exceeded, num: 10, corr: 0.991942
I (34724) esp_radar: time: 500/10, free_heap: 90560, num: 10, rssi: -49, gain: 0/0, wander: 0.008767, jitter: 0.000400, window: 41, begin: 48, end: 19, freq: 82Hz
W (34974) esp_radar: The capacity of subcarrier recording during calibration is exceeded, num: 10, corr: 0.991233
I (34974) esp_radar: time: 509/10, free_heap: 90560, num: 10, rssi: -50, gain: 0/0, wander: 0.009244, jitter: 0.000298, window: 40, begin: 68, end: 38, freq: 78Hz
W (35214) esp_radar: The capacity of subcarrier recording during calibration is exceeded, num: 10, corr: 0.990756
I (35214) esp_radar: time: 501/10, free_heap: 90560, num: 10, rssi: -50, gain: 0/0, wander: 0.008833, jitter: 0.000192, window: 36, begin: 18, end: 54, freq: 71Hz
W (35434) esp_radar: The capacity of subcarrier recording during calibration is exceeded, num: 10, corr: 0.991167

```

Also post training when I start sending csi data I see the following warning 
```
RADAR_DADA,1437,1235444,0.008653,0.004736,0.004296,1,0.001287,0.000751,0.015615,1
W (1235654) algorithm: [169] fail
W (1235654) esp_radar: [509] csi calculate fail
RADAR_DADA,1438,1235894,0.007262,0.004836,0.004296,1,0.000292,0.000751,0.015615,1
I (1235894) mesh_main: [ROOT-2-UNICAST:0][L:2][rtableSize:2]parent:8c:4b:14:14:78:b1 to 8c:4b:14:14:78:b1, [err:0x0, proto:0, tos:0]
I (1236174) esp_radar: time: 511/10, free_heap: 90200, num: 10, rssi: -48, gain: 0/0, wander: 0.008012, jitter: 0.002526, window: 35, begin: 18, end: 53, freq: 68Hz
RADAR_DADA,1439,1236174,0.008012,0.004980,0.004296,1,0.002526,0.000792,0.015615,1
```

ashashik, 2024-08-21:
For 
1. Will this warning affect the wander threshold value accuracy ? I see it shows up within 10 seconds of starting the training . 
2. I just use csi_send program on one node and use console test on the other node . I randomly see this "csi calculate fail" on the node running console test 

```
W (1235654) algorithm: [169] fail
W (1235654) esp_radar: [509] csi calculate fail
```

MacChu0315-Espressif, 2024-08-22:
1. nope, it's merely a warning.
2. which means that there is a problem with the received CSI data, and the algorithm has made a calculation error and thus discarded the data of this frame.

ashashik, 2024-08-26:
Post training I sometimes see the wander threshold is returned as nan . Any way to fix this ? 

RADAR_THRESHOLDS,wander = nan, jitter = 0.010993

so7rry, 2025-03-18:
Hello, I am also encountering this error. The ESP32 that has been burned in works fine on my PC, but when I switch to another laptop, it reports an error and cannot display the CSI waveform chart.