espressif/esp-csi #200: Cant get more than 5 data points

JoaoSandrini, 2024-08-21:
So i tried the examples/get-started/csi_recv_router and i can only get 5 data points, then it stops. I tried setting the ping count to 0 so it pings infinite times, as specified, and it didnt work. I want to be able to continuously transmit through serial and run my own data analisys, but with only 5 data points its impossible.

MacChu0315-Espressif, 2024-08-22:
hi, #172 may help.

JoaoSandrini, 2024-08-22:
I dont think its a problem with baud rate (and its configured correctly, all with a value of 921600), as im able to run the code, get some data points (image link below) and visualize it with examples/get-started/tools/csi_data_read_parse.py. The issue is that it stops getting more data after about 5 (sometimes less) points.

Although im setting ping count to 0 (see image), wich should send infinite packages to the router, it stops after the 4th.

Also, i couldn't run the command lines from the referred issue, i don't know where/how to run it.

[image]
[image]

MacChu0315-Espressif, 2024-08-29:
It failed to reproduce. Can you switch a router or reduce the ping frequency?